ccutrer/balboa_worldwide_app #2: Pinout

los93sol, 2017-06-22:
I'm actually looking to do similar with my tub and trying to figure out the pinout.  Based on your findings I suspect the wifi module is nothing more than a dumb RS232 to Wifi module and that the protocol from the main to the topside is the same as what you're seeing from the Wifi module.  I'm trying to find some documentation on the RJ45 port that Balboa uses on their VS series units.  Does anyone have any info on it or can anyone confirm the wifi module isn't anything special?

ccutrer, 2017-06-23:
The protocol does feel very serial like, and it's infuriating that if one device is connected (or the tub _thinks_ is still connected), another cannot connect, which would also point to masking a single serial port.

There are a few pieces of evidence that it's more than an off-the-shelf serial-to-wifi chipset, though:
 * Balboa has their own MAC address prefixes. This is fairly unusual among embedded devices, where they get MAC address assignments out of the pool of whatever wifi chip they happen to have in them.
 * setting the wifi parameters is done via the normal serial protocol. So either the adapter is watching for that particular sequence to recognize it, or the tub has enough smarts to echo the parameters back to the adapter in whatever the adapter's special protocol is.
 * cloud support. The tub makes an active connection out to balboa's servers to facilitate remote access. I haven't dove into this protocol much, besides recognizing that it's vastly different than the local protocol. And that multiple devices are allowed to connect "remotely" at the same time.

Assuming it is serial to the actual tub, though, none of these special add ons in their adapter should preclude the ability to insert your own serial device instead of the wifi adapter, and control the tub.  It likely would preclude the ability to use the native app (which could be solved by setting up a proxy that pretends to be a tub-at least for local access). Given the flakiness of their wifi adapter, though, it may be a good route.

Unfortunately I'm not much of a low level hardware guy, so don't have the know-how to decipher the pin out. I could probably learn, but any time I get to work on this project would probably be taken up first by deciphering more of the protocol.

los93sol, 2017-06-23:
Great info, my buddy has a tub as well and picked up a TTL->USB converter.  We're able to get data to spew from the tub to putty.  In the process of slamming together a quick app to get the actual stream, but in the meantime, if you can check if there's any web interface exposed by your wifi module that might give a clue as to the proper settings for baud rate, etc. that would be very helpful.  Might be easier to email me directly at los93sol@gmail.com

ccutrer, 2017-06-23:
I'd rather we keep the conversation in a public forum so that if it gets abandoned, at least someone else can see what work we've done.

That said, the following ports are open according to nmap:
 * 110 (normally POP3)
 * 143 (normally IMAP)

When I connect to either of those ports, I just get dead air, even attempting to speak the proper protocols for those ports.

As for actual app - are you comfortable with ruby? I could modify my gem here to work against a serial port pretty easily for you. Which pins are you connecting to which on your adapter?

los93sol, 2017-06-23:
I believe we are getting valid data out of it now using 115200, 8, N, 1....
Here's a sample of some of the messages we were able to dump....

7E-1C-FF-AF-00-00-61-0F-26-00-00-01-00-00-0C-00-00-02-00-00-00-00-00-00-61-00-00-3B-7E
7E-05-BF-06-37-7E
7E-05-BF-07-30-7E
7E-05-10-BF-06-5C-7E
7E-05-BF-06-37-7E-7E-05-BF-07-30-7E
7E-05-10-BF-06-5C-7E
7E-05-BF-06-37-7E-7E-05-BF-07-30-7E
7E-05-10-BF-06-5C-7E
7E-05-BF-06-37-7E-7E-05-BF-07-30-7E
7E-05-10-BF-06-5C-7E
7E-05-BF-06-37-7E-7E-05-BF-07-30-7E
7E-05-10-BF-06-5C-7E
7E-05-BF-06-37-7E-7E-05-BF-07-30-7E
7E-05-10-BF-06-5C-7E
7E-05-BF-06-37-7E-7E-05-BF-07-30-7E
7E-05-10-BF-06-5C-7E
7E-05-BF-06-37-7E-7E-05-BF-07-30-7E
7E-1C-FF-AF-00-00-61-0F-26-00-00-01-00-00-0C-00-00-02-00-00-00-00-00-00-61-00-00-3B-7E

los93sol, 2017-06-23:
I'm working through it now, but so far the message start/message end seems correct, the message length also appears correct so I just need to decode a few of these manually against your documentation to see what all I can glean from it.  I'm stoked that a simple $10 off the shelf adapter seems to be working

ccutrer, 2017-06-23:
Yup, those look super familiar, yet different than any actual messages I have documented so far.

los93sol, 2017-06-23:
We'll have values from another tub on Sunday if things go well, I have a Bullfrog X7 so expect output more similar to what you have with your A7.  Different cable for mine since I don't have the TTL port on mine though so fingers crossed!

los93sol, 2017-06-24:
Is there a trick to being able to send control messages?  We were able to validate out checksums are correct using the sample payload garbled1 posted in another thread here, but so far no luck with a simple toggle lights command.

ccutrer, 2017-06-24:
When using wifi, you can't use any control messages until you've sent one or two messages asking for the configuration. hopefully that's the trick

garbled1, 2017-06-28:
I have a ton of code to do communication with one of these things in my gnhast repo here on github.  The code is in C, but I have most of the routines commented with the protocol as far as I've been able to decode it.

The wifi one spams me like crazy when I connect to it.  4 messages per second.  Does yours do the same over serial?

ccutrer, 2017-06-28:
@garbled1 nice, thanks for the tip! Now I can compare notes against yours. My personal favorites are "BMTR_PANEL_NOCLUE1" and "BMTR_PANEL_NOCLUE2" :). Other than that things are looking pretty familiar. I am curious what an AUX device is? Apparently my tub is fairly simple - I just have two pumps, a circ pump, and one light.

los93sol, 2017-06-29:
We are seeing the same thing, the raw bus is VERRRRRY chatty!

los93sol, 2017-06-29:
Still no luck sending commands, looks like there must be some initialization or translation that the wifi module is doing....or we don't have the connection parameters quite right.  We are able to generate fully valid payloads that mimic what is documented as toggle commands and the commands sent from a real button press, but still no luck

garbled1, 2017-06-30:
@ccutrer 
My tub is pretty simple too.  About a year ago, the controller died, so I bought a generic balboa replacement controller, so it has a ton of features that I don't actually use...

AUX is basically a separate VAC output circuit, 2 amps I think, that you can connect a load to.  For example, lets say you have one of those music controllers, you can wire it into the aux out, and power it via that.

I found that one of them, will return my spa controller's model name, and serial number, encoded in ascii.  I'd really like to see what yours says for that, to see if it's universally decodable.

@los93sol 

So, the wifi connection is the same port type that the panel connects to.  you can literally swap them around and nothing notices.  Maybe you could somehow try to intercept data from the panel and see what it's saying to the spa?

los93sol, 2017-06-30:
@garbled1 We captured some logs early on with the device on a Y adapter and were able to see the panels messages.  We actually have a full capture of the boot process as well and the main board seems to broadcast the results of its memory test on boot too.  Currently we have it connected to a separate port and can still see the panel flooding the bus with 5-6 messages a second, it's crazy how chatty this thing is!

los93sol, 2017-07-09:
After playing with the settings I was using the raw serial bus protocol appears identical to the wifi module, the status messages are 100% identical down to the check byte.  I still cannot get the tub to acknowledge any requests sent to it, and interestingly, where the wifi protocol message type is 0ABF11, they keypad itself sends 11BF11, not sure if this is an encoding issue or not though.  When we were logging initially our status messages weren't 100% aligned to what you see but that turned out to be because we were using UTF8 encoding so some bytes were getting changed on us.

smarty125, 2017-07-27:
Watching this topic...hoping for success

stephb9959, 2017-09-11:
Also watching... I can say that the data is TTL level serial at 115200 8N1 as mentioned. I am using a Photon micro controller to replace the aweful Wifi module. Building a proxy and then will use the Particle cloud to manage the tub.

stephb9959, 2017-09-11:
Quick update...

I have built a simple proxy that intercepts the iOS app and sends to the tub over serial. The goal is to replace the wifi module with something better and more IoT based. 

I can read the codes sent by the tub.  But just like @los93sol, I cannot get the tub to accept any of my commands over serial. And I am just relaying the commands from the iOS app.

If any of you could grab the command generated by the wifi module when you change the set point to 101, i could compare with what the app sends and see if the module does anything to the actual payload.

stephb9959, 2017-09-17:
@garbled1 I am stuck like @los93sol ... All the commands match, CRCs check out. I am even using the commands coming out of the iOS app and they match what I computed. I monitor what I receive and what I send and the tub never reacts.

How are you sending your commands? What kind of hardware are you using? Is there some sort of setup before sending commands?

Thanks for the help

sdelaye, 2017-11-11:
Would be very interested by the response of @garbled1...

los93sol, 2017-11-11:
I actually got it talking, use an rs422 adapter instead

stephb9959, 2017-11-12:
Which RS-422 adapter did you use? I was getting intermittent success with
RS-485 (RS-422 is close to RS-485). I have also a lot of code I have been
writing for all the networking stuff. I will publish this as soon as I get
this working reliably.

los93sol, 2017-11-12:
Whoops, you are correct, I checked my notes and it was an RS-485 adapter that I ended up using

los93sol, 2017-11-12:
On another note, we never accomplished 100% reliability, about 10% of the time messages need to be resent

stephb9959, 2017-11-12:
I don’t know if you discovered this or not but messages cannot just be sent
at random. You need to wait for your time slot. Then you can send
something. Usually you wait for a 06 command and you can either send a 07
if you have nothing to send or whatever command you are wishing to send.

los93sol, 2017-11-12:
Interesting, that explains quite actually, swapping my keypad back and forth between the ports resulted in it getting a new address and I would see phantom messages on the bus that just appeared to be messages from the main board going off into nowhere, I’m just guessing but I would assume that timing logic is implemented in the official WiFi module since you wouldn’t be able to reliably handle that through an app that can connect/disconnect at will

los93sol, 2017-11-12:
Btw, this is the adapter I am using https://www.amazon.com/USB-module-CP2102-Serial-Converter/dp/B01CNV45SG

fabltd, 2019-12-17:
Did any one get this working?

stephb9959, 2019-12-17:
I got some code running on a microcontroler and an rs-485 interface adapter
I bought from sparkfun.com. The project is not complete but I can do a lot
of stuff. Some things I could never figure out. I never tried to run on the
USB adapter.

fabltd, 2019-12-18:
@stephb9959 exellent I have some things working. I cannot get circ pump status to change have you got this working?

stephb9959, 2019-12-19:
Hey Jeremy,

I have not touched this project in a while, after the holidays, I will get
the code out again and see what I can do about that circ pump.

Have you tried grabbing the data packets when you press the keypad to
activate the circ pump?

Cheers

los93sol, 2019-12-19:
I don’t think you do manually control the circ pump.  From what I recall that is automatic based on filter cycles and whether or not the heater/jets are running.

fabltd, 2019-12-19:
Hi

Your correct re control. I am just trying to get accurate status. The
status bit doesn’t change but in the app it shows the pump being on or off.

Regards,
JT

wrote:
>
> Hey Jeremy,
>
> I have not touched this project in a while, after the holidays, I will get
> the code out again and see what I can do about that circ pump.
>
> Have you tried grabbing the data packets when you press the keypad to
> activate the circ pump?
>
> Cheers
>
>
> > @stephb9959 <https://github.com/stephb9959> exellent I have some things
> > working. I cannot get circ pump status to change have you got this
working?
> >
> >
> >
> > <
https://github.com/ccutrer/balboa_worldwide_app/issues/2?email_source=notifications&email_token=AAOF7RCP5NSVVKLBS64ZDNLQZHYPVA5CNFSM4DQLJ63KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEHFS24A#issuecomment-566963568
https://github.com/

los93sol, 2019-12-19:
I’ll double check my code to see what I did to get it, but I have it working in my version.  I don’t have the real WiFi adapter though, I just used a pi and hooked it up that way.

garbled1, 2019-12-20:
Sorry I've been absent from this..  Somehow I missed the updates...

Basically, I'm purely using the wifi adapter.  If you connect to port 4257, it will immediately start spewing data at you.  Mind you, there is a caveat, it can only talk to one thing at a time.  So if your phone is still connected, you won't be able to connect.

I'm actually working on a python module to talk to these, but again, it's for the wifi, not rs4XX...

https://github.com/garbled1/pybalboa

Basically I just send the commands and they work.  Never really had an issue.  Mind you, one problem for some of them, is that you are essentially pushing buttons on a display.  The protocol is truly a raw display.  This means, say you want to change the mode for something from 1 to 5, you have to hit the "change that mode" button 4 times..  so that's 4 commands..

fabltd, 2019-12-20:
Hi 

I have it all working except for the circ pump status. I am using the python module. I have the wifi module. Whats good is the cloud connection uses a different port so you can use this to monitor when you use python to control the tub. 

I am planning to create a Google Home Action for this.

JT

garbled1, 2019-12-20:
What python module are you using?  I couldn't find one which is why I'm writing one..

I can gather circ_pump status, but I don't think you can control it, because it;s theoretically supposed to be always on?

fabltd, 2019-12-20:
Hi

I will find the link your code looks good.

Though you used numpy for the CRC the other version uses the Python CRC
module.

My app shows the CIRC pump switch on and off with the heater.

JT

-- 

Jeremy Turner

Google Cloud Platform Authorised Trainer

Director @fabcontrols

FAB Controls Ltd

Unit 6,St Albans Enterprise Centre

Long Spring, Porters Wood

St Albans, Hertfordshire

AL3 6EN

*Phone:* +44(0) 1727 238 894

Cloud implementation & ‘Internet of Things’ (IoT) solutions provider

garbled1, 2019-12-20:
I could never get any of the base python crc modules to handle the bizzaro CRC format the spa used.  I'd be curious to see other methods.  Mine is basically a direct port of my C code into python.

So there are two types of circulation pump.  For example, I have a 2-pump 2-speed system with heater.  pump 1 is connected to the heater.  When the heater is on, it turns pump 1 on LOW for circulation.
This is NOT considered a "circulation pump" by the system.  A circulation pump is a separate pump wired to a specific port on the control board, which iirc from the docs (not in front of them right now), runs on a different schedule than the rest.

Either way, yes, you can read it off a specific bit, but it's different than all the other status bits, so it's a bit weird.

fabltd, 2019-12-20:
Hi 

This is the example I have working:

https://github.com/plmilord/Hass.io-custom-component-SpaClient/tree/master/custom_components/SpaClient

You just need the spaclient.py .

My circ pump come on and off with the heater. Not sure what this pump is but this is the one the app calls circ pump. 

Do you have a link for the docs?

JT

garbled1, 2019-12-21:
There are no docs for the protocol, this is pure reverse engineering.  What I have are docs for the spa control panel itself, which lists the functions and buttons/menus.   I'm also working on a homeassistant integration, which is why I'm writing the python backend.

Here is the code I use to do the circulation pump:

        if self.circ_pump:
            if data[18] == 0x02:
                self.circ_pump_status = 1
            else:
                self.circ_pump_status = 0

This is the format of that status update:

```
        00 01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16 17 18 19 20 21 22 23
        MS ML MT MT MT XX F1 CT HH MM F2  X  X  X F3 F4 PP  X CP LF MB  X  X  X
        7E 1D FF AF 13  0  0 64  8 2D  0  0  1  0  0  4  0  0  0  0  0  0  0  0

        24 25 26 27 28 29 30
        X  ST X  X  X CB  ME
        0  64 0  0  0  6  7E
```

Thanks for the link to the code.  It looks like your code has a few flags figured out mine might not.. I'll have to dig more.

stephb9959, 2019-12-21:
You could send the button press for all the pumps one by one and see which
one, if any, trigger the circ pump status but.

garbled1, 2019-12-21:
I've got my python module working, released here:
https://github.com/garbled1/pybalboa

I have an issue open, where I'm trying to collect data for some commands that I don't know how to parse the return from just yet.  If anyone has time, I'd appreciate them trying this out on a wifi adapter and appending to my issue with the output.

In theory this could be extended to support a serial connected adapter instead, I don't think it would be super hard, if anyone wants to try, I'd be more than glad to incorporate it.  At this point, I kinda feel like the wifi adapter is the equivalent of an RPI running ser2sock. :)

fabltd, 2019-12-21:
Hi

What commands do you need?

I have a WiFi module.

shaundodimead, 2019-12-21:
I was going to get the Wi-Fi module, how does the rs -232 connect? That little four way fly lead? Is the Wi-Fi module basically just a serial bridge?

shaundodimead, 2019-12-21:
I’m looking to build a home bridge plug in for my spa, using this lib or something similar

shaundodimead, 2019-12-21:
Just  caught up with whole thread - sorry!

garbled1, 2019-12-22:
No problem, welcome to the fun.  If you don't have one yet, I do actually recommend the wifi adapter.  I think it is probably possible to do the serial port, but I don't know that it is well known if it will work or not yet, and the level of hacking will be far higher.  With the wifi adapter and my python library, it should more or less just work.    I believe the library we are all commenting in is a ruby gem?  Also @fabltd has something that I think works.

From looking over the thread, what I believe is that the wifi adapter is just a serial bridge that has a few more commands built in to do things like setup wifi.  From some of the discussions however, I feel there is also a timing component to the messages with pure rs232, so you might be in for a fight there.

@fabltd I need a send panel req with arguments of 2,0  and 4,0  (instead of the usual 0,1) so it would be the message of type [0x0A, 0xBF, 0x22] with byte 5 set to 2 and 7 set to 0 (and then again with byte 5 == 4 and byte 7 == 0)  If you just run my library with the supplied test in __main__.py it will spit them out too.  Basically, I know that one of them is a serial # and model name, but I don't know where exactly the boundaries are in parsing each.  I'm hoping more data makes it more obvious.

stephb9959, 2019-12-22:
FYI, there is no rs-232. It’s rs-485. And it’s basically a bus where you
can read data and send your own commands. Dealing with rs485 is way more
touchy than over WiFi. If you plan on controlling the tub from a PC for
example, WiFi is the way to go I think. In my case, I am building a
microcontroller that uses rs485 and managed over the web.

It’s all in what you want to build.

I’ll eventually post my rs485 code. C based.

los93sol, 2019-12-22:
I can confirm that building your own is not only possible it’s easy and stable.  I’ve had a raspberry pi 3 shoved inside my tub for over a year in a northern climate with temps ranging from -10 to 100f without issue.

ToreBj, 2020-01-06:
Read through this and like to connect serially (RS485). I see the question about which connector and pins mentionend a couple of times, but no answer? Can someone provide an answer ? I have a GL2000 with no wifi module. Thanks!

los93sol, 2020-01-06:
I can confirm using RS485 works just fine.  The protocol is the same directly on the bus.

stephb9959, 2020-01-06:
Hi,

On the wire I have, the pin-out maps to:

Wire:

Yellow is ground

Orange is B

Red is A

Brown is 12v

Now, this does not mean that my cable matches yours. If I were you, I would
use a multimeter and find Ground and +12V. Once you have those, you are
left with A and B. If you wire them in the wrong order, you are just going
to mess the bus, but not damage anything. If you get it right, you will
start seeing data on the bus. Go to any local electronic parts store to
find the actual connector if you want to make your own cable. I found a
plug for $1.50.

Good luck.

adantoft, 2020-04-28:
> I can confirm that building your own is not only possible it’s easy and stable. I’ve had a raspberry pi 3 shoved inside my tub for over a year in a northern climate with temps ranging from -10 to 100f without issue.

@los93sol Are you able to share how you have a raspberry pi interfacing with the tub? I have a 2017 Sundance 880 that I’m looking to integrate with home assistant but hoping to avoid paying $200-$300 for the WiFi module.

closed 2020-05-12 by ccutrer

ten300, 2020-05-13:
Hi, maybe i'm a confused noob but is there a benefit to going through the rs485 port as opposed to the rj45 as this person did? Or does it essentially end up being the same result and either will work? Do all Balboa controllers have rs485 and only some rj45? https://www.olivierhill.ca/archives/72-The-Internet-of-Spas.html

stephb9959, 2020-05-13:
I think you are mixing 2 things here:

-RJ-45 is a connector standard (8 conductors used for ethernet and many
other things)
-rs485 is a communication standard that can be carried over pretty much any
wire and plug. Balboa uses rs485 to send/receive data and they may deliver
that data over any connector: RJ-45, moles, etc. (mine is a 4 pin Molex
connector).

So you have to use rs485, but how you connect is up to the board you are
using

Hope this helps.

ten300, 2020-05-14:
Thanks Stephane I had a doh moment there, of course you are right. I need more coffee!  Having said that the owner of the link I posted has an Arduino sketch that looks usable so I’ll try that route with a nodemcu. My primary goal is to view actual Spa temp (not set temp).  Thanks to you and all for this thread and the work that’s gone into reverse engineering , it’s starting to make sense now

bggardner, 2020-05-25:
I plan on submitting a PR to the docs for this, but I'm going to document the answer properly here first.  The plug connector is a [Molex 430250400](https://www.molex.com/molex/products/part-detail/crimp_housings/0430250400) (or equivalent).  The pinout is:
1. RS-485 B (TX+/RX+/D+)
2. Supply (+12 to 15VDC)
3. Return (0VDC)
4. RS-485 A (TX-/RX-/D-)

Of course, it still doesn't hurt to double-check with a voltmeter as instructed above and now in the README.

cribskip, 2020-06-12:
@stephb9959 I'm very interested in your code as I'd like to create and publish a esp8266 MQTT bridge to the tub. Is your code available for use?

Thanks,
Sascha

[21 more replies]