ccutrer/balboa_worldwide_app #3: Use Putty to see this data?

smarty125, 2017-07-31:
Non programmer here...just a home automation hack.

I am trying to use Putty to see this data stream (to help me better understand the document protocol).
Using Putty I can connect my device to IP 192.168.0.65, port 4257.

I can see an improperly translated data streaming spewing out......my question is, what is the proper character encoding?
I am using the "telnet" (which is TCP ) setting in putty, but specifying port 4257.  

In the translation section of putty, I cannot find a selection that makes the data stream look readable.  Ideas?

ccutrer, 2017-07-31:
You won't be able to us PuTTy. The encoding is packed into raw binary bytes, not ASCII readable characters. The whole point of this project is to understand what those bytes mean.

closed 2017-07-31 by ccutrer

smarty125, 2017-08-13:
Using DockLight Scritping (free version), you can connect to the IP address and port.  The data will be in Hex, and it will coming at you about 2-3 lines (there is no line breaks) per second.