ccutrer/balboa_worldwide_app #4: …More of a “Question” rather than an “Issue”…

smarty125, 2017-08-01:
It looks like some very good “decoding” has been done here to figure out what the hot tub controller is saying, and how to also speak back to it.  I apologize in advance if this is too elementary, but I would like to understand your use case for this work.  

It appears that you run a Ruby program…but what, and how do you interacts with that program?  
How do to you tell the Ruby program to adjust the spa conditions?  How (and to what) does the Ruby program report back spa conditions?

…At the risk of asking too much, how does one go about running this set of .rb files?     ….. (I will do more Googling on how to run ruby, but anything you can say will be appreciated by me and likely many others)…

To better explain why I ask these questions, here is my “use case” that I hope to achieve:

From my home automation and control program (PowerHome), I am hoping that I can run some sort of command line (or write a PH plugin) type program, and have it control and also return spa information.  
That would then allow me to integrate my spa more fully into the rest of my HA system (events, touch screens, reporting and control).  By better understanding your use case, I hope to see how to achieve mine.

Again, thanks for all your fine work.

smarty125, 2017-08-09:
Anyone?

ccutrer, 2017-08-09:
Oops, sorry. I had the notification up, but wasn't able to answer immediately, then forgot about it.

At the moment, these ruby files can't be "run". Each time I try a different command, I modify them a bit. I was waiting until I had a more robust understanding of the protocol before refactoring them into a proper interface usable from an HA controller. But even then, it may be difficult to just integrate into any given HA controller. The spa only allows a single connection at a time, so ideally the controller will keep a persistent connection open, which doesn't really lend itself very well to shelling out to a separate script to get status updates and/or send commands. It's possible, just not as reliable, since establishing a new connection may fail.

Anyhow, the quick answer is that unless you have the skills to look at the protocol documentation and write your own plugin for your chosen HA controller, you'll just have to be patient for this particular project to continue to evolve to the point where a shell-out integration will be possible, and understand that I have many demands on my time and this project is pretty far down the list (I don't even have a true home automation controller yet, and don't plan to get one until I move into a new home - which we haven't even broke ground on).

closed 2020-05-12 by ccutrer