ccutrer/balboa_worldwide_app #13: URI Path RS-485

EricAgostinelli, 2020-08-05:
Where and how exactly do I properly set the uri path for RS-485? 

I've tried many variations, and there's currently one that I've commented out.   Is this the right line to be putting in the URI.Path?  I'm not certain as I'm a not familiar to ruby..

I'm running on a mac, and can't detect the spa.  I've attached a photo of the client error.  

`     
require 'bwa/message'

module BWA

  class Client
    attr_reader :last_status, :last_control_configuration, :last_control_configuration2, :last_filter_configuration

    def initialize(uri)
      uri = URI.parse(uri)
      
      if uri.scheme == 'tcp'
        require 'socket'
        @io = TCPSocket.new(uri.host, uri.port || 4217)
      elsif uri.scheme == 'telnet' || uri.scheme == 'rfc2217'
        require 'net/telnet/rfc2217'
        @io = Net::Telnet::RFC2217.new("Host" => uri.host, "Port" => uri.port || 23, "baud" => 115200)
        @queue = []
      else
        require 'serialport'
        ##@io = SerialPort.open(uri.path:"/dev/tty.usbserial-AR0JVV52", "baud" => 115200)
        @io = SerialPort.open(uri.path, "baud" => 115200)
        @queue = []
      end
      @buffer = ""
    end

`

[image]

ccutrer, 2020-08-10:
You just pass it as the first argument:

`bwa_client /dev/tty.usbserial-AR0JVV52`

Auto-detection only works for the Balboa wifi module.

EricAgostinelli, 2020-10-02:
Sorry for such a delay on this.   I still had some issues, perhaps could you share a screenshot of how you have it set up?    My next attempt is running it separately on a pi.  To see if that works

ccutrer, 2020-10-04:
`bwa_mqtt_bridge mqtts://mqtt.cutrer.network/ /dev/ttyHotTub`

closed 2020-12-21 by ccutrer