ccutrer/balboa_worldwide_app #16: M100_235 v52.0 Status Message Error

richard-otto-obrien, 2020-10-31:
I'm seeing the following error log when running on a RPi 3 with a same model USB RS485 adapter (DSD Tech SH-U11) as you specify.

```
okt 31 18:03:44 Raspi-HotTub bwa_mqtt_bridge[677]: #<Thread:0x00d611d8@/var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/bin/bwa_mqtt_bridge:20 run> terminated with exception (report_on_exception is true):
okt 31 18:03:44 Raspi-HotTub bwa_mqtt_bridge[677]: /var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/lib/bwa/message.rb:59:in `parse': Unrecognized data length (32) for message BWA::Messages::Status (BWA::InvalidMessage)
```

I'm wondering whether this is something tied to my SSID being one that is not mentioned in your documentation/wiki. It is the following: `M100_235 v52.0`

My Balboa controller is a BP200G3 and I connected my RPi & RS485 adapter not via a Y-cable but directly to the second jumper connector intended for the Wifi controller I believe.

richard-otto-obrien, 2020-10-31:
If I run `hexdump /dev/ttyUSB0` I get the following (see below). This appears to me to be messages. I've yet to try to decipher these to see if they fit messages from the protocol.

```
0000000 3e00 7e7e 1005 06bf 7e5c 077e bf10 0011
*
0000040 0000 7e3e 057e bf10 5c06 7e7e 1007 11bf
0000050 0000 7e3e 207e afff 0013 4e03 020e 2800
0000060 0301 0403 0000 0000 0000 0200 4e02 0000
0000070 1e00 0000 7e69 057e bf10 5c06 7e7e 0700
0000080 bf10 0011 3e00 7e7e 1005 06bf 7e5c 077e
*
0000100 bf10 7e11 077e bf10 0011 3e00 7e7e 1005
0000110 06bf 7e5c 077e bf10 0011 3e00 7e7e 1005
*
00001d0 06bf 7e5c 077e bf10 0011 3e00 7e7e ff20
00001e0 13af 0300 0e4e 0002 0128 0303 0004 0000
00001f0 0000 0000 0202 004e 0000 001e 6900 7e7e
0000200 1005 06bf 7e5c 077e bf10 0011 3e00 7e7e
*
00002e0 ff20 13af 0300 0e4e 0002 0128 0303 0004
00002f0 0000 0000 0000 0202 004e 0000 001e 6900
0000300 7e7e 1005 06bf 7e5c 077e bf10 0011 3e00
*
0000390 7e7e fe05 00bf 7eac 057e bf10 5c06 7e7e
00003a0 1007 11bf 0000 7e3e 057e bf10 5c06 7e7e
*
00003d0 1007 11bf 0000 7e3e 207e afff 0013 4e03
00003e0 020e 2800 0301 0403 0000 0000 0000 0200
00003f0 4e02 0000 1e00 0000 7e69 057e bf10 5c06
0000400 7e7e 1007 11bf 0000 7e3e 057e bf10 5c06
*
00004d0 7e7e 1007 11bf 0000 7e3e 207e afff 0013
00004e0 4e03 020e 2800 0301 0403 0000 0000 0000
00004f0 0200 4e02 0000 1e00 0000 7e69 057e bf10
0000500 5c06 7e7e 1007 11bf 0000 7e3e 057e bf10
*
00005d0 5c06 7e7e 1007 11bf 0000 7e3e 207e afff
00005e0 0013 4e03 020e 2800 0301 0403 0000 0000
00005f0 0000 0200 4e02 0000 1e00 0000 7e69 057e
0000600 bf10 5c06 7e7e 1007 11bf 0000 7e3e 057e
*
00006e0 bffe ac00 7e7e 1005 06bf 7e5c 077e bf10
00006f0 0011 3e00 7e7e ff20 13af 0300 0e4e 0002
0000700 0128 0303 0004 0000 0000 0000 0202 004e
0000710 0000 001e 6900 7e7e 1005 06bf 7e5c 077e
0000720 bf10 0011 3e00 7e7e 1005 06bf 7e5c 077e
*
00007d0 bf10 0011 3e00 7e7e ff20 13af 0300 0e4e
00007e0 0002 0128 0303 0004 0000 0000 0000 0202
00007f0 004e 0000 001e 6900 7e7e 1005 06bf 7e5c
0000800 077e bf10 0011 3e00 7e7e 1005 06bf 7e5c
*
00008d0 077e bf10 0011 3e00 7e7e ff20 13af 0300
00008e0 0e4e 0002 0128 0303 0004 0000 0000 0000
00008f0 0202 004e 0000 001e 6900 7e7e 1005 06bf
0000900 7e5c 077e bf10 0011 3e00 7e7e 1005 06bf
*
00009d0 7e5c 077e bf10 0011 3e00 7e7e ff20 13af
00009e0 0300 0e4e 0002 0128 0303 0004 0000 0000
00009f0 0000 0202 004e 0000 001e 6900 7e7e 1005
0000a00 06bf 7e5c 077e bf10 0011 3e00 7e7e 1005
*
0000a30 06bf 7e5c 077e bf10 0011 3e00 7e7e fe05
0000a40 00bf 7eac 057e bf10 5c06 7e7e 1007 11bf
0000a50 0000 7e3e 057e bf10 5c06 7e7e 1007 11bf
*
0000ad0 0000 7e3e 207e afff 0013 4e03 020e 2800
0000ae0 0301 0403 0000 0000 0000 0200 4e02 0000
0000af0 1e00 0000 7e69 057e bf10 5c06 7e7e 1007
0000b00 11bf 0000 7e3e 057e bf10 5c06 7e7e 1007
*
0000bd0 11bf 0000 7e3e 207e afff 0013 4e03 020e
0000be0 2800 0301 0403 0000 0000 0000 0200 4e02
0000bf0 0000 1e00 0000 7e69 057e bf10 5c06 7e7e
0000c00 1007 11bf 0000 7e3e 057e bf10 5c06 7e7e
*
```

bggardner, 2020-10-31:
I believe this is a bug in the [parser](https://github.com/ccutrer/balboa_worldwide_app/blob/master/lib/bwa/message.rb#L59), as it assumes all messages of the same type/class are the same length.  The wiki shows status messages can be different lengths, depending on the version.  The fix probably involves changing `Message::MESSAGE_LENGTH` to an array and checking if the received length is in the array, or avoiding the length check altogether.

richard-otto-obrien, 2020-10-31:
@bggardner I'm not at all familiar with Ruby so I've got no idea how to implement the message length check. For the time being I removed the length check altogether.

Like this:
```
        if not klass
        #  raise InvalidMessage.new("Unrecognized data length (#{length}) for message #{klass}", data) unless length - 5 == klass::MESSAGE_LENGTH
        #else
          klass = Unrecognized
        end
```
At some point I might revisit this and see if I can implement the check.

closed 2020-12-21 by ccutrer