ccutrer/balboa_worldwide_app #17: TypeError in poll (client.rb)

thomas2403, 2020-11-18:
I'm using the WiFi module. Port 4217 (client.rb line 11) is not working, so I specified port 4257. Then I get the following error:
 ```
#<Thread:0x000055bad5654920@/var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/bin/bwa_mqtt_bridge:20 run> terminated with exception (report_on_exception is true):

Traceback (most recent call last):
	6: from /var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/bin/bwa_mqtt_bridge:21:in `block in initialize'
	5: from /var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/bin/bwa_mqtt_bridge:21:in `loop'
	4: from /var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/bin/bwa_mqtt_bridge:23:in `block (2 levels) in initialize'
	3: from /var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/lib/bwa/client.rb:26:in `poll'
	2: from /var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/lib/bwa/client.rb:26:in `loop'
	1: from /var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/lib/bwa/client.rb:33:in `block in poll'
/var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/lib/bwa/client.rb:33:in `send': no implicit conversion of Symbol into String (TypeError)
 ```
My Balboa controller is a BP2100G0.

globaltopmedia, 2020-11-30:
Got the same issue (both on the port, and then the TypeError)

Output straight from netcat
Connection to ************ 4257 port [tcp/*] succeeded!
)>??
)>?J?Ic?hh~~!??#n~~%??
)>?J?Ic?hh~~%??
)>?J?Ic?hh~~%??
)>?J?Ic?hh~~!??#n~~%??
)>?J?Ib?hh~~%??
)>?J?Ib?hh~~%??
)>?J?Ib?hh~~!??#n~~%??
)>?J?Ib?hh~~%??
)>?J?Ib?hh~~%??

thomas2403, 2020-12-02:
I found a solution. Line 39 in client.rb says:
 ```
@buffer.concat(@io.send(method, 64 * 1024))
 ```
The method send is a common method in ruby. Unfortunately TCPSocket overwrites this method, that's why we get the error using the WiFi module. Just change the line into:
 ```
@buffer.concat(@io.__send__(method, 64 * 1024))
 ```
This calls the original method. It is working for me, but I can not test it with the other types (Telnet, SerialPort). 
Maybe someone who is using SerialPort can test this and post the result here. 

BTW, after I got this working, I ran into the Status Message Error https://github.com/ccutrer/balboa_worldwide_app/issues/16
So this project might need some work anyway.

ccutrer, 2020-12-21:
Yeah... my bad. I totally should have caught this. I'll release an updated gem shortly with the fix for __send__

closed 2020-12-21 by ccutrer