ccutrer/balboa_worldwide_app #18: Process takes ~100% cpu

konsulten, 2020-12-09:
Current platform where Im running it is a raspberry pi 3 +. Directly after starting it takes up 100% cpu and stays there. Question is if there is some while loop without proper sleep or something missing?

konsulten, 2020-12-11:
It seems like its in the poll function in client, if read is used it does not block until data is available. and thus spinning on reading 0 bytes. Added a if @buffer.length = 0 sleep (0.1) end which seems to have done the trick

ccutrer, 2020-12-21:
Were you using the master branch when you reported this, and using RS-485? If not, you should either do that, or update to the 1.2.0 release which I just pushed. There were problems with the old serial library that it would return immediately when there was no data. I switch the serial library backend and it fixed the problem. It's been running for months at my house with no issues; I apologize for not getting around to releasing a new version.

closed 2020-12-21 by ccutrer

konsulten, 2020-12-21:
I were using the master branch when I reported  this. Will check if it was
any updates with your 1.2.0 rel.

ccutrer, 2020-12-21:
Serial or WiFi?

konsulten, 2020-12-21:
Oh, sorry. Rs-485