ccutrer/balboa_worldwide_app #22: How to debug this app?

agillis, 2021-03-07:
I have a Pi attached to my hot tub with the suggested RS-485 USB adapter. 

I start bwa_mqtt_bridge and it runs with no errors. 

It registers with my MQTT server but all values are NULL.

I did a hexdump /dev/ttyUSB0 and a large amount of data seems to be going by.

What else can I troubleshoot this? Is there any way to turn on debugging so I can see what's happening?

ccutrer, 2021-03-07:
Most likely you have your positive and negative wires swapped connecting your RS-485 adapter

agillis, 2021-03-08:
I do. When I had them the other way (like the picture on readme)  I was getting an error. I'll swap them back and post the error tomorrow. Thanks.

agillis, 2021-03-14:
I have it connected like you picture now.

This is the error I get

/usr/local/bin/bwa_mqtt_bridge mqtt://192.168.1.20/ /dev/ttyUSB0

discarding invalid data prior to message 10bf065c7e
wrote 7e050abf04777e
#<Thread:0x01521d50@/var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/bin/bwa_mqtt_bridge:20 run> terminated with exception (report_on_exception is true):
Traceback (most recent call last):
        6: from /var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/bin/bwa_mqtt_bridge:21:in `block in initialize'
        5: from /var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/bin/bwa_mqtt_bridge:21:in `loop'
        4: from /var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/bin/bwa_mqtt_bridge:23:in `block (2 levels) in initialize'
        3: from /var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/lib/bwa/client.rb:26:in `poll'
        2: from /var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/lib/bwa/client.rb:26:in `loop'
        1: from /var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/lib/bwa/client.rb:27:in `block in poll'
/var/lib/gems/2.5.0/gems/balboa_worldwide_app-1.1.2/lib/bwa/message.rb:59:in `parse': Unrecognized data length (32) for message BWA::Messages::Status (BWA::InvalidMessage)

It does connect to my broker but does not send any data.

How can I debug what the problem is?

ccutrer, 2021-03-15:
Ooh interesting. You have it connected correctly now, but your status message is even longer than any of the rest of us have gotten yet. I’ll do a commit to allow the longer message.

bggardner, 2021-03-15:
@ccutrer I was going to say let's not bother with checking message lengths (or throw warnings instead of errors for messages exceeding an expected _minimum_ length), but this way it helps us discover new configurations.

@agillis Once you're up and running, please contribute to the Status Update and/or Examples sections of the [wiki](https://github.com/ccutrer/balboa_worldwide_app/wiki), or at least post your SSID here so we can track the differences.  Thanks!

ccutrer, 2021-03-15:
1.2.1 just pushed to rubygems with the slightly more lenient status message length

closed 2021-03-15 by ccutrer