ccutrer/balboa_worldwide_app #28: Registration

kw217, 2021-03-31:
This app is currently a bit unreliable on my spa, and I think it's because we are trying to use the same channel as the panel unit.

I'm thinking of adding registration to the app (in client.rb), using the logic in https://github.com/rsrawley/spaControl/blob/master/spa.js#L283 . Does that sound OK, and has anyone attempted this already?

bggardner, 2021-03-31:
I've done it in [Python](https://github.com/bggardner/pybalboa/blob/master/pybalboa/clients.py), following the [Channel Assignment protocol](https://github.com/ccutrer/balboa_worldwide_app/wiki#channel-assignment-request).  Not sure what @rsrawley is doing with the "default" `preferredChannel: "11"`, but the channel should only be supplied by the Main Board (not assumed or hard-coded).

ccutrer, 2021-04-01:
Oh wow, I hadn't even seen all that channel assignment stuff in the wiki. Yes, definitely I'd be happy to merge it if you write it (and I'll try it out with my spa).

rsrawley, 2021-04-01:
If I recall correctly, I believe my intention with preferredChannel was to avoid increasing the assigned channel by one every time I ran my code when testing.  I still need to fix the case when the spa loses power and restarts, as my program does not check if its assigned channel is still active.  At the moment, I have to restart the program if I restart the spa.

kw217, 2021-04-01:
@rsrawley I don't have that issue, because I have it running on a Pi powered off the Balboa main board - so if the spa loses power so does the program. I'll see if I can do something about that case.

@ccutrer cool, thanks - I'll see what I can do :-) I have a few days off over the weekend, if I have a following wind I should get to it!