ccutrer/balboa_worldwide_app #29: OpenHab autodiscovery not compatible with Homie version 4.0.0

TheNetStriker, 2021-04-21:
I've just setup your app and I noticed that my spa does not automatically appear in OpenHab as a new Thing. I found out that your app is sending Homie version 4.0.0 which is not yet compatible with OpenHab. So I simply changed the version to 3.0.0 and the spa did appear automatically in the Thing autodiscovery and the items also work.

As far as I've seen there is no need for version 4.0 with the items you defined. Can you change the version in your code too to fix this problem?

ccutrer, 2021-04-21:
OpenHAB does in fact support Homie 4.0: https://github.com/openhab/openhab-addons/blob/main/bundles/org.openhab.binding.mqtt.homie/src/main/java/org/openhab/binding/mqtt/homie/internal/discovery/Homie300Discovery.java#L72, and my tub has been added to openhab via auto discovery. I have noticed that OpenHAB can be finicky and not always show Homie devices right away.

closed 2021-04-21 by ccutrer

TheNetStriker, 2021-04-21:
Thanks for the quick reply. I found the following in my log:

openhab Found homie device. But version v4.0.0 is out of range.

I noticed that you changed the version code from v4.0.0 to 4.0.0. I guess that this fixed the issue, but the version with this change is not yet published.

ccutrer, 2021-04-21:
Ah yes, I did have that problem because I accidentally had the 'v'. I'll publish a new version of the gem.