ccutrer/balboa_worldwide_app #30: Blower not working with Balboa Verona Alpha spa.

TheNetStriker, 2021-04-21:
After setting up your spa bridge I noticed that the blower command does not work with my spa. (Balboa Verona Alpha) This spa only has one speed for the blower and the pumps. (On or Off)

I noticed that the blower topic is configured as an integer value, but the value that was sent by the bridge was "false". I did take a look at the code and found a workarround that works for me:

In bin/bwa_mqtt_bridge I changed the line
publish_attribute("spa/blower", message.blower) if @bwa.last_control_configuration2&.blower.to_i != 0
to
publish_attribute("spa/blower", message.blower ? 1 : 0) if @bwa.last_control_configuration2&.blower.to_i != 0

And in the lib/bwa/client.rb file I change the line
times = (desired - last_status.blower) % (last_control_configuration2.blower + 1)
to
times = (desired - (last_status.blower ? 1 : 0)) % (last_control_configuration2.blower + 1)

I guess this fix only works for spa's with only one speed setting. I'm not too familiar with Ruby and this code, so I'm hoping that you can find a better solution for this.

There is also a strange behaviour with the pumps. If I set the pumps to 1 they turn on, so far so good. If I then set the pumps back to 0 nothing happens. But If I then change the value back to 1 the pumps turn off and the number automatically changes to 0. This also happens if I set the value to 2.

I guess the best solution would be to detect if the blower and pumps only support one speed and set the Homie device datatype to boolean instead of integer. Can you please take a look at this?

ccutrer, 2021-04-21:
I'd be happy to update the Homie metadata to only do booleans, but I need to know how to detect that that is the case first, and I can't do that with my spa. You should probably look in the wiki at the protocol documents, and see if you can determine if there's a bit in any of the status or configuration messages to help you identify this condition.

TheNetStriker, 2021-04-21:
@ccutrer As far as I can see the max speed is already detected. The format of the blower and pumps are all set to 0:1. This seems to be set in this method:

  def publish_pump(i, speeds)
    publish("spa/pump#{i}/$name", "Pump #{i} speed")
    publish("spa/pump#{i}/$datatype", "integer")
    publish("spa/pump#{i}/$format", "0:#{speeds}")
    publish("spa/pump#{i}/$settable", "true")
    subscribe("spa/pump#{i}/set")

    @things << "pump#{i}"
    publish_nodes
  end

Couldn't you use this to set the device datatype to boolean?

TheNetStriker, 2021-04-22:
@ccutrer I now found out what is causing the strange behaviour on the spa pumps. I placed some debug code in the set_pump method and found out that the pumps report the speed as 0 or 2. Here is what I found in the variables:

last_status.pumps[i - 1] = 0 or 2 for off and on
last_control_configuration2.pumps[i - 1] = always 1

The 0 or 2 speed is also reported in the Status message, but the number 2 is blocked by the format 0:1 of the Homie specification in OpenHab. Also the code to calculate the times variable does not work with this. (When sending a 0 it always set's the times variable to 0) This is why I can only switch the pump by sending another 1 or a 2. I'm currently thinking about a good solution to support both speed configurations. Maybe you have a good idea?

I also did take a look at the blower protocol and as far as I can see the blower can only be only on or off. I guess in that case the Homie datatype of the blower could be permanently changed to boolean. I already tested this and it works fine with two other small changes. I could send you a pull request for this.

TheNetStriker, 2021-04-23:
I've just had an idea and created a Fork of the repository with my changes. This is working pretty well for me now:

https://github.com/TheNetStriker/balboa_worldwide_app/commit/fa7bb6292f933f82a7597fe5de4e66cef75c27d7

Maybe you could take a look at it and implement something like this into your repository.

ccutrer, 2021-11-10:
https://github.com/TheNetStriker/balboa_worldwide_app/tree/PumpSingleSpeedFix has been merged into master.

closed 2021-11-10 by ccutrer