ccutrer/balboa_worldwide_app #31: Chromazon3 not working

digwa-ing, 2021-08-01:
I can't toggle the lights. It is probably because the lights are part of the chromazon3 system. 
Is there a way if controlling them?

digwa-ing, 2021-08-01:
I am trying to get the codes for controlling the LEDs. Unfortunately I cannot connect to TCP Port 4257 using netcat. Am I missing something?

digwa-ing, 2021-08-12:
Good news: the Chromazon3-System can be controlled by the serial controller (including the WiFi module).

I am working on figuring out the messages for the Chromazon3 System.

The Systems sends out a status every second:

`7E 18 FF AF 32 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 7E`	light is off
`7E 18 FF AF 32 01 01 00 57 FF FF FF 01 00 57 FF FF FF 01 00 57 FF FF FF 4E 7E`		light is white (first programmed color)
`7E 18 FF AF 32 01 02 00 57 FF 00 00 02 00 57 FF 00 00 02 00 57 FF 00 00 F5 7E`	light is red (second programmed color)

The Message Type 32 seems to be a status update for the Chromazon3-System

The following commands can be used to switch the lights:
`7E 0D FF AF 31 00 00 00 00 00 00 00 00 BB 7E`	turn lights off
`7E 0D FF AF 31 11 00 00 00 00 00 00 00 9F 7E`	change all zones to white
`7E 0D FF AF 31 01 00 01 00 00 00 00 00 81 7E`	change all zones to 2nd programmed color
`7E 0D FF AF 31 01 00 02 00 00 00 00 00 FA 7E`	change all zones to 2nd programmed color
`7E 0D FF AF 31 01 00 03 00 00 00 00 00 D3 7E`	change all zones to 3rd programmed color
`7E 0D FF AF 31 02 00 00 00 00 00 00 00 9D 7E`	change all zones to color-rotation

=> byte 0 seems to be the mode (solid color vs. color-rotation)
=> byte 1: when byte1=0: select color out of programmed colors

I will continue my investigations.

bggardner, 2021-08-12:
Great job!  Feel free to update the [wiki](https://github.com/ccutrer/balboa_worldwide_app/wiki) with your findings as you go.

ccutrer, 2021-12-11:
@digwa-ing: any additional info on this? looking at the status messages, I think I can see three sets of RGB values, which would correspond to zones A, B, and C (which I know there are 3 zones from my two seconds of googling to get to https://balboawater.com/CZ_ST3; I've never heard of Chromazon 3 before this issue). If you can get more detail on the messages, I'd be more than happy to implement them in the MQTT bridge.

Given that I don't have a Chromazon 3 system to play with, you'll have to help me with the details, though. For example, you say programmed color... on the control panel can you choose preset colors with a color picker? Given that the status message seems to be full RGB, I'm kinda surprised there are presets at all, at least at the protocol level. So capturing how a preset is saved would be good. And then changing one zone at a time, so we can identify what byte is "control this zone" in the control messages, as well as the order of the zones in the status messages.

digwa-ing, 2022-08-23:
@ccutrer sorry for the delay. I did some testing and figured out the following:
Message type 31 is for sending commands to Chromazon3
Turn all lights off: 7E 0D FF AF 31 00 00 00 00 00 00 00 00 BB 7E
White light on all zones: 7E 0D FF AF 31 11 00 00 00 00 00 00 00 9F 7E
Red Light on all zones: 7E 0D FF AF 31 01 00 01 00 00 00 00 00 81 7E

There are programmable slots for pre-defined colors.
This selects the first color for all zones: 7E 0D FF AF 31 01 00 01 00 00 00 00 00 81 7E
second color for all zones: 7E 0D FF AF 31 01 00 02 00 00 00 00 00 fa 7E  
third color for all zones: 7E 0D FF AF 31 01 00 03 00 00 00 00 00 d3 7E 
=>third bit seems to be the pre-defined color-slot to select

There is also the option to let the colors continuously change / fade:
7E 0D FF AF 31 02 00 00 00 00 00 00 00 9D 7E
=> the first bit seems to be the mode (01 for pre-defined color, 02 for continuous colors)

The commands above work for changing colors on all zones (ABC). There are codes for changing the colors on the zones individually but I didn't get to document them yet.

Every second, the system sends out the current status of the lights. e.g.:
lights are off: 7E 18 FF AF 32 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 7E
lights are white: 7E 18 FF AF 32 01 01 00 57 FF FF FF 01 00 57 FF FF FF 01 00 57 FF FF FF 4E 7E
lights are red: 7E 18 FF AF 32 01 02 00 57 FF 00 00 02 00 57 FF 00 00 02 00 57 FF 00 00 F5 7E

digwa-ing, 2022-08-28:
I finished my investigations.
Here is a video of my test: https://youtu.be/eVvw7I1__ko
Here are the data recordings:
[rec3.txt](https://github.com/ccutrer/balboa_worldwide_app/files/9440035/rec3.txt)
I filtered out the action calls and captioned them:
[commands.txt](https://github.com/ccutrer/balboa_worldwide_app/files/9440038/commands.txt)

Here is my conclusion:
Bytes
Call Cromazon3-Action: 7e 0d ff af 31 (Type Code= 31)

Byte5 (1)
00 Turn lights off
01 Change color (color-number in Byte7: 0=color1, 1=color2, …) if lights are of, they will turn on
02 Color Wheel Light Show if Byte7=00, Party Light Show if Byte7=01, if Byte7=3: Relax/Calm Light Show
03 dim lights
04  change fading speed for Light Shows (speed (hex) in sconds in Byte12). If a solid color is selected, changing the fading speed will activate the Color Wheel Light Show
09 edit color (Byte6: Zone; Byte7: Color-Number; Byte8: Red, Byte9: Green; Byte10: Blue)
11 Turn lights on (last set color will be used)

Byte6 (2)
00: All Zones (Group 1)
01: Group 2
81: Zone A
82: Zone B
84: Zone C
When Programming Colors (Byte5=09): don't use 00, use 81 instead

Byte7 (3)
Color-number (if Byte5=01) (0…7)
Light-Show if Byte5=02:
	00: Color Wheel Light Show
	01: Party Light Show
	03: Relax/Calm Light Show

Byte8 (4)
When editing stored Color (Byte5: 09): Red (0…255)

Byte9 (5)
When editing stored Color (Byte5: 09): Green (0…255)

Byte10 (6)
When editing stored Color (Byte5: 09): Blue (0…255)

Byte11 (7)
Brightnes 0-100% in hex, 0% turns the lights off

Byte12 (8)
Fading speed for Light Shows in seconds
0 pauses the fade

For your plugin I suggest to keep it simple: No Zones, No Lightshows: only all Lights on, all Lights off, change color and dim the lights. It would look like this:
Turn On (optional, since it will also turn on when a color is being selected or the lights ar being dimmed; this will turn on all LEDs to the last selected color)
7e0dffaf3111000000000000009f7e

Program Color7: (you can store 8 colors (0...7), I suggest only editing color7; Colors from 00 to FF; RR=Red; GG=Green; BB=Blue)
7e0dffaf31098107RRGGBB0000e37e

Select Color7: (the stored color has to be selected in order to light up)
7e0dffaf310100070000000000777e

Dim: (PP is the percentage, possible values are 100%, 87% (x57), 74% (x4a), 61% (x3d), 48% (x30), 35% (x23), 22% (x16), 9% (x9), 0% (x0) = off)

Turn off
echo '7e0dffaf310000000000000000bb7e' | xxd -r -p | nc 192.168.178.178 4257

After editing, the color must be reselected!

I hope, the information is sufficient. Please let me know, if you need any more information. It would problby also be possible to get the current status of the lights, but my investigations didn't go that far yet. They could probably be figured out from the rec3.txt-file.