ccutrer/balboa_worldwide_app #32: Gem can't be installed on Raspberry

JochenKoehler69, 2021-08-04:
Hi,

I just tried to install your gem on an Raspberry Pi 2 Model B Rev 1.1 - but got this error message:
```
root@loxberry:/opt/loxberry# gem install balboa_worldwide_app
Building native extensions. This could take a while...
ERROR:  Error installing balboa_worldwide_app:
        ERROR: Failed to build gem native extension.

    current directory: /var/lib/gems/2.5.0/gems/digest-crc-0.6.4/ext/digest
/usr/bin/ruby2.5 -rrubygems /usr/share/rubygems-integration/all/gems/rake-12.3.1/exe/rake RUBYARCHDIR=/var/lib/gems/2.5.0/extensions/arm-linux/2.5.0/digest-crc-0.6.4 RUBYLIBDIR=/var/lib/gems/2.5.0/extensions/arm-linux/2.5.0/digest-crc-0.6.4
/usr/bin/ruby2.5: No such file or directory -- /usr/share/rubygems-integration/all/gems/rake-12.3.1/exe/rake (LoadError)

rake failed, exit code 1

Gem files will remain installed in /var/lib/gems/2.5.0/gems/digest-crc-0.6.4 for inspection.
Results logged to /var/lib/gems/2.5.0/extensions/arm-linux/2.5.0/digest-crc-0.6.4/gem_make.out
```
Do you have any advice what to do?
It's a Raspbian distribution:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Thanks for your help!

JochenKoehler69, 2021-08-04:
I investigated a bit more and installed "rake" separately - then I get this error message:

```
root@loxberry:/opt/loxberry# gem install balboa_worldwide_app
Building native extensions. This could take a while...
ERROR:  Error installing balboa_worldwide_app:
        ERROR: Failed to build gem native extension.

    current directory: /var/lib/gems/2.5.0/gems/digest-crc-0.6.4/ext/digest
/usr/bin/ruby2.5 -rrubygems /var/lib/gems/2.5.0/gems/rake-13.0.6/exe/rake RUBYARCHDIR=/var/lib/gems/2.5.0/extensions/arm-linux/2.5.0/digest-crc-0.6.4 RUBYLIBDIR=/var/lib/gems/2.5.0/extensions/arm-linux/2.5.0/digest-crc-0.6.4
mkmf.rb can't find header files for ruby at /usr/lib/ruby/include/ruby.h

rake failed, exit code 1

Gem files will remain installed in /var/lib/gems/2.5.0/gems/digest-crc-0.6.4 for inspection.
Results logged to /var/lib/gems/2.5.0/extensions/arm-linux/2.5.0/digest-crc-0.6.4/gem_make.out
```

JochenKoehler69, 2021-08-04:
I found the solution on my own :-)
To be able to build the gem native extensions an additional package "ruby-dev" has to be installed:

sudo apt-get install ruby-dev.

With this installed, the installation of balboa_worldwide_app worked fine.

Thanks

closed 2021-08-04 by JochenKoehler69

EricAgostinelli, 2021-08-04:
This is great to know!   I too have had a terrible time trying to get this to work on a Pi.   I will have to try this myself.  Thanks @JochenKoehler69