ccutrer/balboa_worldwide_app #38: Patch to add Filter Cycles and Hold functionality

BowenMarmot, 2021-10-30:
The attached patch adds the ability to get and set filter cycle 1 and 2 and the "Hold" toggle. 
There are also a couple of bug fixes that I needed to make it work.
  - format of the URL (bwa_client)
  - removal of the homie format entry for currenttemperature since this is irrelevant as the value could be well out of the defined range. (e.g. when heating from a fresh fill). In my case, with openhab, this prevented the display of any value outside the format. (bwa_mqtt_bridge)
  - update of default TCP port to 4257 as defined on the Wiki (client.rb)
  - removal of extraneous "&" on when serializing current temperature (status.rb)

I don't have a github fork and don't have much familiarity with git hub, hence a patch instead of a pull request.

[filtercycles_patch.txt](https://github.com/ccutrer/balboa_worldwide_app/files/7447451/filtercycles_patch.txt)

JochenKoehler69, 2021-10-31:
@BowenMarmot Could you please give me a hint how to apply your patch - I'm not familiar with "patch" ... Thanks!

JochenKoehler69, 2021-10-31:
Seems to work well :-) - Just testing it for a while before doing a pull request.
Thanks a lot :-)

ccutrer, 2021-11-10:
PR has been merged.

closed 2021-11-10 by ccutrer