ccutrer/balboa_worldwide_app #40: GS 502 PCB only RJ45 connectors

fwittenb, 2021-11-04:
Hi,
Thanks for. this great writeup - most current spa's have a new Balboa PCB board where the 4 wire connector is removed in total. I did do some testing waiting on a protocol analyser to finalise it. But someone might all-ready have solved it but i did not find it in the documentation, where only the 4 wire connection is documented.

[image of GS502](https://www.google.com/search?q=balboa+controller+board&rlz=1C9BKJA_enUS810US810&hl=en-GB&prmd=ismxvn&source=lnms&tbm=isch&sa=X&ved=2ahUKEwi61L3D__7zAhWR6qQKHbqMCaEQ_AUoAXoECAIQAQ&biw=1112&bih=719&dpr=2#imgrc=bxK6N-sS2oPltM)

Below so far my measurements and findings i'm not seeing a 12V wire hence my ask and need to further investigate before connecting components and a pi to the spa. 

```
#| interface side | voltage | function |
#|----------------|---------|----------|
#|  pin 1         |  5.15v     vcc ?
#|  pin 2         |  0v        gnd ?
#|  pin 3         |  0.14v     copy of pin 4 or noice ? 
#|  pin 4         |  0.25v     clock - nice symetic signal 0-1
#|  pin 6         |  3.23v
#|  pin 7         |  0v        gnd ?
#|  pin 8         |  4.98v
```
Thanks Frans

bggardner, 2021-11-04:
The RJ-45 plug uses a completely different protocol than the Molex connectors (not RS-485).  This issue is more or less covered in #14.

closed 2021-11-10 by ccutrer

netmindz, 2022-02-06:
@fwittenb - Not sure how similar the GS range is to the GL, but as similar age, they might be more similar that the BS range.
Might be worth following https://github.com/netmindz/balboa_GL_ML_spa_control