ccutrer/balboa_worldwide_app #43: Control heating

JochenKoehler69, 2021-11-11:
I would like to prevent the spa from heating when energy prices are too high (My contract has varying prices on an hourly basis) and force heating (in certain ranges) when energy is available from the PV.
What is the best way to do this (of course I can change the settemperature - but maybe there is a better way ...)?
With other words: I have seen the status "Heatingmode" and "Hold" but I do not know the meaning of these.

ccutrer, 2021-11-11:
There are a few things you can play with. Temperature range is essentially two separate set points. I think this is meant to be used as a "vacation mode". 

Heating mode can be set to either ready, or rest. When in rest mode, the tub will run the circ pump less often in order to check the current temperature, and I believe it allows the temperature to deviate more from the set point in order to save energy. If you manually start one of the other pumps, it's a signal to the tub "I'm here!", and it will temporarily go into "ready in rest" mode for some period of time.

Hold mode, if my memory serves, is only tangentially related to temperature control and is instead meant for maintenance. It holds _any_ pump (including circ) from running temporarily, so you can say remove and clean your filters without worry that the pump will kick on while they're not in place.

If what you're trying to accomplish is to keep the tub as close to your set point as possible, but prioritize heating while solar output is high, you can take two approaches. The lighter weight approach is to just set heating mode to rest while solar isn't producing, and ready otherwise. If you're okay with wider temperature swings while solar isn't producing, then yes I would ignore the heating mode logic and just directly control the set point yourself, lowering it say 5° while solar isn't producing, and then restoring (possibly an extra 1-2° higher than you normally would) while it is producing.

JochenKoehler69, 2021-11-21:
Heating mode is working. Thanks for the hint.

JochenKoehler69, 2021-12-17:
I made some experience with the heating mode and realized that the actual heating is not reacting well when heating mode is switched from "rest" to "ready". It keeps heating off. I have to switch on lights for example - then the spa is waking up again and turns heating on! Did anyone make similar observations? Any hints on improving this behavior?

closed 2023-06-16 by ccutrer