ccutrer/balboa_worldwide_app #45: filterx settings throw NoMethodError

jshank, 2021-11-16:
I've not had any luck setting any of the new filter options via MQTT. They all throw `NoMethodError` and then Balboa MQTT Bridge restarts.

**Setting filter1hour to 12**
```
W, Balboa MQTT Bridge running (version 1.3.0)
W, from mqtt: "ready" at homie/bwa/spa/heatingmode/set
W, from mqtt: "12" at homie/bwa/spa/filter1hour/set
/usr/local/bundle/gems/balboa_worldwide_app-1.3.0/lib/bwa/client.rb:67:in `send_message': undefined method `src=' for "\n\xBF#\f\x00\x03\x00\b\x00\x00\x0F":String (NoMethodError)
   from /usr/local/bundle/gems/balboa_worldwide_app-1.3.0/lib/bwa/client.rb:200:in `set_filtercycles'
   from /usr/local/bundle/gems/balboa_worldwide_app-1.3.0/bin/bwa_mqtt_bridge:186:in `block in initialize'
   from /usr/local/bundle/gems/mqtt-0.5.0/lib/mqtt/client.rb:405:in `block in get'
   from /usr/local/bundle/gems/mqtt-0.5.0/lib/mqtt/client.rb:437:in `block in get_packet'
   from /usr/local/bundle/gems/mqtt-0.5.0/lib/mqtt/client.rb:435:in `loop'
   from /usr/local/bundle/gems/mqtt-0.5.0/lib/mqtt/client.rb:435:in `get_packet'
   from /usr/local/bundle/gems/mqtt-0.5.0/lib/mqtt/client.rb:404:in `get'
   from /usr/local/bundle/gems/balboa_worldwide_app-1.3.0/bin/bwa_mqtt_bridge:147:in `initialize'
   from /usr/local/bundle/gems/balboa_worldwide_app-1.3.0/bin/bwa_mqtt_bridge:602:in `new'
   from /usr/local/bundle/gems/balboa_worldwide_app-1.3.0/bin/bwa_mqtt_bridge:602:in `<top (required)>'
   from /usr/local/bundle/bin/bwa_mqtt_bridge:23:in `load'
   from /usr/local/bundle/bin/bwa_mqtt_bridge:23:in `<main>'
```

**Setting filter1minute to 10**
```
W, from mqtt: "10" at homie/bwa/spa/filter1minute/set
/usr/local/bundle/gems/balboa_worldwide_app-1.3.0/lib/bwa/client.rb:67:in `send_message': undefined method `src=' for "\n\xBF#\r\n\x03\x00\b\x00\x00\x0F":String (NoMethodError)
   from /usr/local/bundle/gems/balboa_worldwide_app-1.3.0/lib/bwa/client.rb:200:in `set_filtercycles'
   from /usr/local/bundle/gems/balboa_worldwide_app-1.3.0/bin/bwa_mqtt_bridge:186:in `block in initialize'
   from /usr/local/bundle/gems/mqtt-0.5.0/lib/mqtt/client.rb:405:in `block in get'
   from /usr/local/bundle/gems/mqtt-0.5.0/lib/mqtt/client.rb:437:in `block in get_packet'
   from /usr/local/bundle/gems/mqtt-0.5.0/lib/mqtt/client.rb:435:in `loop'
   from /usr/local/bundle/gems/mqtt-0.5.0/lib/mqtt/client.rb:435:in `get_packet'
   from /usr/local/bundle/gems/mqtt-0.5.0/lib/mqtt/client.rb:404:in `get'
   from /usr/local/bundle/gems/balboa_worldwide_app-1.3.0/bin/bwa_mqtt_bridge:147:in `initialize'
   from /usr/local/bundle/gems/balboa_worldwide_app-1.3.0/bin/bwa_mqtt_bridge:602:in `new'
   from /usr/local/bundle/gems/balboa_worldwide_app-1.3.0/bin/bwa_mqtt_bridge:602:in `<top (required)>'
   from /usr/local/bundle/bin/bwa_mqtt_bridge:23:in `load'
   from /usr/local/bundle/bin/bwa_mqtt_bridge:23:in `<main>'
W, Balboa MQTT Bridge running (version 1.3.0)
```

BowenMarmot, 2021-11-16:
You are absolutely correct!  It seems when the last round of patches were merged someone had done some refactoring of the way that messages were sent to the spa that didn't work with my code.  I have pulled the latest version and can replicate the issue.  

I spent a bit of time understanding the refactoring and updating the the filter code and with some minimal testing, it is now working for me.  I will do some more testing tomorrow and package it up into a pull request. 

Thanks so much for the report.

jshank, 2021-11-17:
@BowenMarmot I pulled ccutrer/balboa_worldwide_app#46 and it's working great on my tub. Thanks!

BowenMarmot, 2021-11-18:
Thanks, that is great to hear!

Just as a heads up since I gather from your commits you are using HomeAssistant, there is one race condition I have discovered and resolved in my own branch. When an automation sends updates to a number of the filter cycle values in rapid succession it can cause a second (or third or ..) message to be sent to the spa before the previous, new values, get updated from the spa.  The somewhat inelegant but functional "fix" is to slow down the calls to the message with a "sleep" in the client.rb.  I am going to hold off submitting another PR for the moment in case there are other edge cases that come up in testing/use.

ccutrer, 2021-12-10:
this is working on `main`; I'll release a new gem in the next day or two

closed 2021-12-10 by ccutrer