ccutrer/balboa_worldwide_app #47: "Can't Find Spa!"

EricAgostinelli, 2021-11-16:
Noticing there has been  a lot of activity lately, I figured I'd re-visit.

Current setup:
Pi Zero-W with 485 adapter.

Can confirm communication is on /dev/USB0 Bus.  Previously I had the Pi Set up with ser2net to communicate to a Crestron processor. 

Issue:

When attempting to run the program it can't detect the spa for 485 communication.  

Attached is a screenshot of .service and the status terminal
[image]
[image]

I'm stumped. I gave up on this for about a year and went a different path, however I'd still like this functioning regardless.

jshank, 2021-11-17:
Start simple and see if you're getting valid data on `/dev/ttyUSB0` 

`hexdump /dev/ttyUSB0`

You may have to `sudo` that command if your user doesn't have access to `/dev/ttyUSB0`

ctrl-c to terminate the dump.

You should see something like below:
```
0000000 bf10 5c06 7e7e 1007 11bf 0000 7e3e 057e
*
00000e0 bf10 5c06 7e7e 1007 11bf 0000 7e3e 1d7e
00000f0 afff 0013 6003 050c 0400 003c 1c00 0000
0000100 0002 0000 0100 6320 0000 f102 7e7e 1005
0000110 06bf 7e5c 077e bf10 0011 3e00 7e7e 1005
```

The biggest hint that you have valid data is a bunch of hex character `7e` in the output. The balboa protocol uses `7e` as a [delimiter](https://github.com/ccutrer/balboa_worldwide_app/wiki#message-structure) for almost every message.

 - If you aren't getting any data, you have problems with your connection or device.
 - If you are getting data but there aren't any hex character `7e`, try swapping your A/B connections on your RS-485 device and hexdump again.
 - If you are now seeing `7e`, fire up bwa and you should be good

Do report back and we can troubleshoot more.

EricAgostinelli, 2021-11-18:
[image]
Still when opening(Execute in Terminal)  BWA_MQTT_Bridge it shows Could Not Find Spa and then quickly closes the terminal window.

BowenMarmot, 2021-11-18:
I don't have answers for you but a couple more troubleshooting steps to try and gather more information.

1) Not sure how you installed the gem but it relies on another serialport gem. Run the following:

    gem list |grep -i serial

    You should get a result like:
    ccutrer-serialport (1.0.0)

2) See what the serialport parameters are for /dev/ttyUSB0. The code defaults to baud 115200 and maybe this is different. Run the command below and share the results. 
     sudo stty -F /dev/ttyUSB0

3) Try running the gem manually in DEBUG mode. Make sure that nothing is running (systemctl stop balboa_worldwide_app) and then run the following.  Not sure it will provide any further details but worth a try.
   LOG_LEVEL=DEBUG /usr/local/bin/bwa_mqtt_bridge mqtt://0.0.0.0/ /dev/ttyUSB0

EricAgostinelli, 2021-11-18:
[image]

Looks good so far...   And I have some results below.   Also, what appeared when terminating the dump.

But I don't think I'm getting the full result.   
Also, when trying to execute in terminal the bwa_mqtt_bridge is still get the could not find spa and then quickly closes the sesion. 

See below for the results as requested. 

pi@Spa-Pi:~ $ LOG_LEVEL=DEBUG /usr/local/bin/bwa_mqtt_bridge mqtt://0.0.0.0/ /dev/ttyUSB0
I,   to spa: #<BWA::Messages::ConfigurationRequest>
D,   to mqtt: $homie: 4.0.0
D,   to mqtt: $name: BWA Spa
D,   to mqtt: $state: init
D,   to mqtt: $nodes: spa
D,   to mqtt: spa/$name: BWA Spa
D,   to mqtt: spa/$type: spa
D,   to mqtt: spa/$properties: hold,priming,heatingmode,temperaturescale,24htime,heating,temperaturerange,currenttemperature,settemperature,filter1,filter2
D,   to mqtt: spa/hold/$name: In Hold mode
D,   to mqtt: spa/hold/$datatype: boolean
D,   to mqtt: spa/hold/$settable: true
D,   to mqtt: spa/priming/$name: Is the pump priming
D,   to mqtt: spa/priming/$datatype: boolean
D,   to mqtt: spa/heatingmode/$name: Current heating mode
D,   to mqtt: spa/heatingmode/$datatype: enum
D,   to mqtt: spa/heatingmode/$format: ready,rest,ready_in_rest
D,   to mqtt: spa/heatingmode/$settable: true
D,   to mqtt: spa/temperaturescale/$name: Temperature scale
D,   to mqtt: spa/temperaturescale/$datatype: enum
D,   to mqtt: spa/temperaturescale/$format: fahrenheit,celsius
D,   to mqtt: spa/temperaturescale/$settable: true
D,   to mqtt: spa/24htime/$name: Clock is 24 hour time
D,   to mqtt: spa/24htime/$datatype: boolean
D,   to mqtt: spa/24htime/$settable: true
D,   to mqtt: spa/heating/$name: Heater is currently running
D,   to mqtt: spa/heating/$datatype: boolean
D,   to mqtt: spa/temperaturerange/$name: Current temperature range
D,   to mqtt: spa/temperaturerange/$datatype: enum
D,   to mqtt: spa/temperaturerange/$format: high,low
D,   to mqtt: spa/temperaturerange/$settable: true
D,   to mqtt: spa/currenttemperature/$name: Current temperature
D,   to mqtt: spa/currenttemperature/$datatype: float
D,   to mqtt: spa/settemperature/$name: Set Temperature
D,   to mqtt: spa/settemperature/$datatype: float
D,   to mqtt: spa/settemperature/$settable: true
D,   to mqtt: spa/filter1/$name: Filter cycle 1 is currently running
D,   to mqtt: spa/filter1/$datatype: boolean
D,   to mqtt: spa/filter2/$name: Filter cycle 2 is currently running
D,   to mqtt: spa/filter2/$datatype: boolean
D,   to mqtt: spa/filter1hour/$name: Filter Cycle 1 Start Hour
D,   to mqtt: spa/filter1hour/$datatype: integer
D,   to mqtt: spa/filter1hour/$format: 0:24
D,   to mqtt: spa/filter1hour/$settable: true
D,   to mqtt: spa/filter1minute/$name: Filter Cycle 1 Start Minutes
D,   to mqtt: spa/filter1minute/$datatype: integer
D,   to mqtt: spa/filter1minute/$format: 0:59
D,   to mqtt: spa/filter1minute/$settable: true
D,   to mqtt: spa/filter1durationhours/$name: Filter Cycle 1 Duration Hours
D,   to mqtt: spa/filter1durationhours/$datatype: integer
D,   to mqtt: spa/filter1durationhours/$format: 0:24
D,   to mqtt: spa/filter1durationhours/$settable: true
D,   to mqtt: spa/filter1durationminutes/$name: Filter Cycle 1 Duration Minutes
D,   to mqtt: spa/filter1durationminutes/$datatype: integer
D,   to mqtt: spa/filter1durationminutes/$format: 0:59
D,   to mqtt: spa/filter1durationminutes/$settable: true
D,   to mqtt: spa/filter2enabled/$name: Filter Cycle 2 Enabled
D,   to mqtt: spa/filter2enabled/$datatype: boolean
D,   to mqtt: spa/filter2enabled/$settable: true
D,   to mqtt: spa/filter2hour/$name: Filter Cycle 2 Start Hour
D,   to mqtt: spa/filter2hour/$datatype: integer
[... 50 lines cut ...]
D,  read: 7e 07 10 bf 11 00 00 3e 7e
D, from spa: #<BWA::Messages::ToggleItem 0>
D,  read: 7e 07 10 bf 11 00 00 3e 7e
D, from spa: #<BWA::Messages::ToggleItem 0>
D,  read: 7e 07 10 bf 11 00 00 3e 7e
D, from spa: #<BWA::Messages::ToggleItem 0>
D,  read: 7e 07 10 bf 11 00 00 3e 7e
D, from spa: #<BWA::Messages::ToggleItem 0>
D,  read: 7e 07 10 bf 11 00 00 3e 7e
D, from spa: #<BWA::Messages::ToggleItem 0>

Below is what appeared when terminating the command

^C/var/lib/gems/2.3.0/gems/mqtt-0.5.0/lib/mqtt/client.rb:436:in `pop': Interrupt
	from /var/lib/gems/2.3.0/gems/mqtt-0.5.0/lib/mqtt/client.rb:436:in `block in get_packet'
	from /var/lib/gems/2.3.0/gems/mqtt-0.5.0/lib/mqtt/client.rb:435:in `loop'
	from /var/lib/gems/2.3.0/gems/mqtt-0.5.0/lib/mqtt/client.rb:435:in `get_packet'
	from /var/lib/gems/2.3.0/gems/mqtt-0.5.0/lib/mqtt/client.rb:404:in `get'
	from /var/lib/gems/2.3.0/gems/balboa_worldwide_app-1.3.0/bin/bwa_mqtt_bridge:149:in `initialize'
	from /var/lib/gems/2.3.0/gems/balboa_worldwide_app-1.3.0/bin/bwa_mqtt_bridge:614:in `new'
	from /var/lib/gems/2.3.0/gems/balboa_worldwide_app-1.3.0/bin/bwa_mqtt_bridge:614:in `<top (required)>'
	from /usr/local/bin/bwa_mqtt_bridge:22:in `load'
	from /usr/local/bin/bwa_mqtt_bridge:22:in `<main>'

pi@Spa-Pi:~ $

BowenMarmot, 2021-11-18:
That debug run shows that it is actually working.  You can see that it is reading from the spa and writing to the spa.

As well, when I take another look at the systemctl status message you posted initially it shows no errors and running properly.

Let's go back to basics.

1) How do you KNOW that this is not working? Have you tried changing a value in MQTT when it is running? If so what happens?  
2) What is the exact command line you are issuing when you get the "spa not found" error in the terminal?  
If it is just "/usr/local/bin/bwa_mqtt_bridge" then the error is expected because you need to pass in your MQTT Broker and your Serial Port in the same way as the ExecStart line in your systemctl service and in the DEBUG line.

EricAgostinelli, 2021-11-19:
BowenMarmot,  I think you maybe on to what I'm doing wrong.  lol
I've been pressing "Execute in Terminal"  when opening the "bwa_mqtt_bridge" shortcut that I put on my desktop . (see pic) 
[image]

The shortcut is from the "/usr/local/bin/bwa_mqtt_bridge". (see pic)
[image]

With this being said, I was under the impression that by executing that file it would load the program as well as the mqtt path.   It's seeming this is incorrect? since that's when I receive the same error when doing so from terminal.  (see pic)
[image]
 
By running /usr/local/bin/bwa_mqtt_bridge mqtt://0.0.0.0/ /dev/ttyUSB0, I receive a response and it just holds until I cancel. (see pic).
[image]

jshank, 2021-11-19:
That makes sense. Running just `/usr/local/bin/bwa_mqtt_bridge` without a target serial device will try to use the auto discovery logic for the Balboa Wi-Fi Module 50350, which, I assume you don't have. Adding the proper parameter is pointing to your MQTT server and USB RS-485 adapter. Sounds like everything is working correctly. There's no need to manually launch the `bwa_mqtt_bridge` command as the system service will run it automatically for you.

Just to confirm, do you have Mosquitto or another MQTT broker installed locally on the Pi?

```
gem install balboa_worldwide_app
sudo curl https://github.com/ccutrer/balboa_worldwide_app/raw/master/contrib/bwa_mqtt_bridge.service -L -o /etc/systemd/system/bwa_mqtt_bridge.service
<modify the file to pass the correct URI to your MQTT server, and path to RS-485 device or hostname/IP for WiFi>
sudo systemctl enable bwa_mqtt_bridge
sudo systemctl start bwa_mqtt_bridge
```

ccutrer, 2021-12-11:
Closing as stale, and it looks like it was properly communicating, just a misunderstanding of how to run it in the background.

closed 2021-12-11 by ccutrer