ccutrer/balboa_worldwide_app #50: handling of MQTT URI with special character

squirtbrnr, 2021-11-30:
I have a username/password that contains special characters.  When I setup the MQTT URI as such: `mqtt://username:pa##word@hostname.domain:port` the app starts, throws an error asking if the URI is actually a URI, then stops unexpectedly.  I am running the app in a docker container created by @jshank and he determined his container image properly passes the environment variable to the app, but the app is having difficulty parsing it.  I have tried passing the URI in the environment variable using single quotes and also tried escaping the `#` but neither worked.  This is the error in the docker container log:

```
/usr/local/lib/ruby/2.6.0/uri/rfc3986_parser.rb:67:in `split': bad URI(is not URI?): "mqtt://username:pa##word@ip:port" (URI::InvalidURIError)
	from /usr/local/lib/ruby/2.6.0/uri/rfc3986_parser.rb:73:in `parse'
	from /usr/local/lib/ruby/2.6.0/uri/common.rb:234:in `parse'
	from /usr/local/bundle/gems/mqtt-0.5.0/lib/mqtt/client.rb:562:in `parse_uri'
	from /usr/local/bundle/gems/mqtt-0.5.0/lib/mqtt/client.rb:152:in `initialize'
	from /usr/local/bundle/gems/balboa_worldwide_app-1.2.5/bin/bwa_mqtt_bridge:13:in `new'
	from /usr/local/bundle/gems/balboa_worldwide_app-1.2.5/bin/bwa_mqtt_bridge:13:in `initialize'
	from /usr/local/bundle/gems/balboa_worldwide_app-1.2.5/bin/bwa_mqtt_bridge:283:in `new'
	from /usr/local/bundle/gems/balboa_worldwide_app-1.2.5/bin/bwa_mqtt_bridge:283:in `<top (required)>'
	from /usr/local/bundle/bin/bwa_mqtt_bridge:23:in `load'	
	from /usr/local/bundle/bin/bwa_mqtt_bridge:23:in `<main>'
```

I have not tried this again on the 1.3.0 app version.  Ultimately I ended up creating a new username/password that did not contain special characters and that worked.

ccutrer, 2021-11-30:
Did you URI escape your password?

squirtbrnr, 2021-11-30:
sorry I don't understand the question.  What I'm saying is the docker container was passing the string of the URI I set in the environment variable and treating it as a whole string.  But when the app inside the docker container attempted to read the URI, it failed with the above error.  the double `##` in the password was just an example.  the actual password in my URI only has one `#`.

jshank, 2021-11-30:
I think I get it, BWA needs the URI pre-encoded for special characters (# being one of them) like this:

`mqtt://username:pa%23%23word@ip:port /dev/hottub`

You have to use [percent encoding](https://en.wikipedia.org/wiki/Percent-encoding#Reserved_characters) for the reserved characters.

jshank, 2021-11-30:
@ccutrer - Would there be any impact to adding URI.escape [parsing to the mqtt_uri section](https://github.com/ccutrer/balboa_worldwide_app/blob/82c9882db8e1510933ef8c351bfb27a3c95ee028/bin/bwa_mqtt_bridge#L595)?

e.g.
`595: mqtt_uri = URI.escape(ARGV.shift)` 

Seems to address the issue with a quick test
```ruby
# irb
irb(main):001:0> URI.escape("mqtt://balboa:pa##word@10.1.10.2")
=> "mqtt://balboa:pa%23%23word@10.1.10.2"
irb(main):002:0> exit
```

ccutrer, 2021-11-30:
Yes, there will be a major impact. It will break anyone that's properly escaping their password already. I absolutely refuse to do it. Proper encoding/escaping/decoding of strings in general and URIs in particular is a pet peeve of mine. We can definitely make mention in the readme to "be sure to properly URI encode your password in the URI if you use authentication, and have special characters. <example>"

closed 2021-11-30 by ccutrer