ccutrer/balboa_worldwide_app #52: Blower not working

Chrone86, 2021-12-10:
Hi
If i try to set blower (homie/bwa/spa/blower/set), the service stopps with the following error:
"W, Balboa MQTT Bridge running (version 1.3.0)
W, from mqtt: "" at homie/bwa/spa/blower/set
Traceback (most recent call last):
        12: from /usr/local/bin/bwa_mqtt_bridge:23:in `<main>'
        11: from /usr/local/bin/bwa_mqtt_bridge:23:in `load'
        10: from /usr/local/share/gems/gems/balboa_worldwide_app-1.3.0/bin/bwa_mqtt_bridge:614:in `<top (required)>'
         9: from /usr/local/share/gems/gems/balboa_worldwide_app-1.3.0/bin/bwa_mqtt_bridge:614:in `new'
         8: from /usr/local/share/gems/gems/balboa_worldwide_app-1.3.0/bin/bwa_mqtt_bridge:149:in `initialize'
         7: from /usr/local/share/gems/gems/mqtt-0.5.0/lib/mqtt/client.rb:404:in `get'
         6: from /usr/local/share/gems/gems/mqtt-0.5.0/lib/mqtt/client.rb:435:in `get_packet'
         5: from /usr/local/share/gems/gems/mqtt-0.5.0/lib/mqtt/client.rb:435:in `loop'
         4: from /usr/local/share/gems/gems/mqtt-0.5.0/lib/mqtt/client.rb:437:in `block in get_packet'
         3: from /usr/local/share/gems/gems/mqtt-0.5.0/lib/mqtt/client.rb:405:in `block in get'
         2: from /usr/local/share/gems/gems/balboa_worldwide_app-1.3.0/bin/bwa_mqtt_bridge:184:in `block in initialize'
         1: from /usr/local/share/gems/gems/balboa_worldwide_app-1.3.0/lib/bwa/client.rb:145:in `set_blower'
/usr/local/share/gems/gems/balboa_worldwide_app-1.3.0/lib/bwa/client.rb:145:in `-': false can't be coerced into Integer (TypeError)"

ccutrer, 2021-12-10:
I'm pretty sure I already fixed this yesterday. Do you know how many speeds your blower has? Can your run the code from main, or do I need to release a new version of the gem for you to test it?

Chrone86, 2021-12-10:
The blower has only one speed, as my pumps have. I have no idea how to run the code from main, i installed it last time with gem

ccutrer, 2021-12-10:
Okay, I just released v2.0.0 of the gem. 🤞 it works for you. Note that it is a major release, as some of the properties have been renamed.

Chrone86, 2021-12-11:
I tried to install it with: gem install balboa_worldwide_app

_Building native extensions. This could take a while...
ERROR:  Error installing balboa_worldwide_app:
        ERROR: Failed to build gem native extension.

    current directory: /usr/local/share/gems/gems/json-2.6.1/ext/json/ext/generator
/usr/bin/ruby -r ./siteconf20211211-2961059-wrumyr.rb extconf.rb
mkmf.rb can't find header files for ruby at /usr/share/include/ruby.h

extconf failed, exit code 1

Gem files will remain installed in /usr/local/share/gems/gems/json-2.6.1 for inspection.
Results logged to /usr/local/lib64/gems/ruby/json-2.6.1/gem_make.out_

As I am not a ruby specialist, i'm not sure if something with your realesase is not working or the failure is on my side. Maybe you have an idea?

Chrone86, 2021-12-11:
If i try to start this release, i get the following error message:

Traceback (most recent call last):
        7: from /usr/local/bin/bwa_mqtt_bridge:23:in `<main>'
        6: from /usr/local/bin/bwa_mqtt_bridge:23:in `load'
        5: from /usr/local/share/gems/gems/balboa_worldwide_app-2.0.0/exe/bwa_mqtt_bridge:389:in `<top (required)>'
        4: from /usr/local/share/gems/gems/balboa_worldwide_app-2.0.0/exe/bwa_mqtt_bridge:389:in `new'
        3: from /usr/local/share/gems/gems/balboa_worldwide_app-2.0.0/lib/bwa/client.rb:48:in `initialize'
        2: from /usr/local/share/gems/gems/balboa_worldwide_app-2.0.0/lib/bwa/client.rb:48:in `new'
        1: from /usr/local/share/gems/gems/ccutrer-serialport-1.0.0/lib/ccutrer/serial_port.rb:7:in `initialize'
/usr/local/share/gems/gems/ccutrer-serialport-1.0.0/lib/ccutrer/serial_port.rb:7:in `initialize': No such file or directory @ rb_sysopen - 192.168.100.141 (Errno::ENOENT)

ccutrer, 2021-12-11:
Hmm. What version of Ruby are you using? You definitely got the new code running finally.

Chrone86, 2021-12-11:
ruby 2.5.9p229 (2021-04-05 revision 67939) [x86_64-linux]

Chrone86, 2021-12-11:
If I try with rvm ruby 3.0.3, i get this error:
/usr/local/rvm/gems/ruby-3.0.3/gems/balboa_worldwide_app-2.0.0/exe/bwa_mqtt_bridge:206:in `block in publish_basic_attributes': undefined method `zero?' for false:FalseClass (NoMethodError)
        from /usr/local/rvm/gems/ruby-3.0.3/gems/homie-mqtt-1.6.0/lib/mqtt/homie/device.rb:57:in `block in node'
        from /usr/local/rvm/gems/ruby-3.0.3/gems/homie-mqtt-1.6.0/lib/mqtt/homie/device.rb:145:in `init'
        from /usr/local/rvm/gems/ruby-3.0.3/gems/homie-mqtt-1.6.0/lib/mqtt/homie/device.rb:53:in `node'
        from /usr/local/rvm/gems/ruby-3.0.3/gems/balboa_worldwide_app-2.0.0/exe/bwa_mqtt_bridge:141:in `publish_basic_attributes'
        from /usr/local/rvm/gems/ruby-3.0.3/gems/balboa_worldwide_app-2.0.0/exe/bwa_mqtt_bridge:51:in `initialize'
        from /usr/local/rvm/gems/ruby-3.0.3/gems/balboa_worldwide_app-2.0.0/exe/bwa_mqtt_bridge:394:in `new'
        from /usr/local/rvm/gems/ruby-3.0.3/gems/balboa_worldwide_app-2.0.0/exe/bwa_mqtt_bridge:394:in `<main>'

ccutrer, 2021-12-11:
👍

I'll look as soon as I can. The 3.0 stack trace looks like it might be a bug occurring because I don't have a blower to actually test with, but I should be able to figure it out.

The 2.5 stack trace is probably a Ruby 2.5-specific bug that I can also track down. I assume you're running on a Raspberry Pi? They really need to update their system Ruby. That's specifically why I (try to) maintain compatibility with 2.5, but it's so ancient!!

Anyhow, expect an update from me by Monday afternoon at the latest.

Chrone86, 2021-12-11:
Thank you very much. I also tried with 2.7.5 and got the same error as on 3.0.3
No, i'm running it on centos 8.

ccutrer, 2021-12-11:
v2.0.1 was just published, which fixes both the Ruby 2.5 issue and reading blower status. Let me know how it goes. I haven't yet tested with Ruby 3.0, just waiting for it to compile.

closed 2021-12-11 by ccutrer

ccutrer, 2021-12-11:
Yup, looks like it's working just fine under 3.0 as well.

Chrone86, 2021-12-12:
thank you!
Something still doesn't work as expected.

If i set 0,1 or 2 nothing happens. If i set true or false on "homie/bwa/spa/blower/set" then i get the following exception:
W, Balboa MQTT Bridge running (version 2.0.1)
Traceback (most recent call last):
        7: from /usr/local/rvm/gems/ruby-2.7.5/gems/homie-mqtt-1.6.0/lib/mqtt/homie/device.rb:105:in `block (2 levels) in publish'
        6: from /usr/local/rvm/gems/ruby-2.7.5/gems/mqtt-ccutrer-1.0.1/lib/mqtt/client.rb:444:in `get'
        5: from /usr/local/rvm/gems/ruby-2.7.5/gems/mqtt-ccutrer-1.0.1/lib/mqtt/client.rb:444:in `loop'
        4: from /usr/local/rvm/gems/ruby-2.7.5/gems/mqtt-ccutrer-1.0.1/lib/mqtt/client.rb:458:in `block in get'
        3: from /usr/local/rvm/gems/ruby-2.7.5/gems/homie-mqtt-1.6.0/lib/mqtt/homie/device.rb:116:in `block (3 levels) in publish'
        2: from /usr/local/rvm/gems/ruby-2.7.5/gems/homie-mqtt-1.6.0/lib/mqtt/homie/property.rb:171:in `set'
        1: from /usr/local/rvm/gems/ruby-2.7.5/gems/balboa_worldwide_app-2.0.1/exe/bwa_mqtt_bridge:223:in `block (2 levels) in publish_basic_attributes'
/usr/local/rvm/gems/ruby-2.7.5/gems/balboa_worldwide_app-2.0.1/lib/bwa/client.rb:193:in `blower=': undefined method `-' for true:TrueClass (NoMethodError)

ccutrer, 2021-12-12:
🤦 . easy fix. but yeah, your single speed blower (and pumps if you have them) no longer accept 0, 1, and 2. Just true/false/toggle.

ccutrer, 2021-12-12:
... but that is a good idea to still allow integer speeds. 2.0.4 allows that, with more intuitive conversions.

Chrone86, 2021-12-12:
Blower works now like a charm. Thank you for your support. 
At the moment i have to toggle the pumps with homie/bwa/spa/pump1/set 1 for on and off. Would it be possible to activate the pumps with true and deactivate with false? My pumps are single speed.

ccutrer, 2021-12-12:
What's in MQTT at homie/bwa/spa/pump1/$format?

Chrone86, 2021-12-12:
the $format doesnt exist..
Looks like this for my balboa
[image]

Chrone86, 2021-12-12:
By the way, I made some advertise for your plugin on a german loxone site https://loxwiki.atlassian.net/wiki/spaces/LOXBERRY/pages/1636466747/Balboa+Spa

Chrone86, 2021-12-13:
i just disabled "convert booleans". 
The pump1 starts and stop now with true and 1. 

Blower (true/false works):
[image]

Light1 (true/false works):
[image]

Pump1 (true and 1 works for start and stop)
Start with true:
[image]

Send false:
[image]

Stop with true:
[image]

commit cd89f952 (2021-12-11): read all blower speeds in status message

fixes #52
commit ee243dd7 (2021-12-12): fix setting state for single-speed blower

fixes #52 (again)