ccutrer/balboa_worldwide_app #53: Unable to turn off pumps

jshank, 2022-01-08:
I've lost the ability to turn pumps back off. It looks like bwa is crashing without error (logging set to debug). It seems to be related to the switch to pumps as true false instead of 0,1,2.

MQTT Logs
```
homie/bwa/$homie 4.0.0
homie/bwa/$name BWA Link
homie/bwa/$state ready
homie/bwa/$nodes spa,filter-cycle1,filter-cycle2
homie/bwa/spa/$name Hot Tub
homie/bwa/spa/$type CL501X1
homie/bwa/spa/$properties hold,priming,heating-mode,temperature-scale,twenty-four-hour-time,heating,temperature-range,current-temperature,target-temperature,circulation-pump,pump1,pump2,pump3,light1
homie/bwa/spa/priming false
homie/bwa/spa/priming/$name Priming
homie/bwa/spa/priming/$datatype boolean
homie/bwa/spa/heating false
homie/bwa/spa/heating/$name Heating
homie/bwa/spa/heating/$datatype boolean
homie/bwa/spa/pump1 true
homie/bwa/spa/pump1/$name Pump 1
homie/bwa/spa/pump1/$datatype boolean
homie/bwa/spa/pump1/$settable true
homie/bwa/spa/pump2 false
homie/bwa/spa/pump2/$name Pump 2
homie/bwa/spa/pump2/$datatype boolean
homie/bwa/spa/pump2/$settable true
homie/bwa/spa/pump3 false
homie/bwa/spa/pump3/$name Pump 3
homie/bwa/spa/pump3/$datatype boolean
homie/bwa/spa/pump3/$settable true
homie/bwa/spa/light1 false
homie/bwa/spa/light1/$name Lights
homie/bwa/spa/light1/$datatype boolean
homie/bwa/spa/light1/$settable true
homie/bwa/spa/hold false
homie/bwa/spa/hold/$name Hold
homie/bwa/spa/hold/$datatype boolean
homie/bwa/spa/hold/$settable true
homie/bwa/spa/heating-mode ready
homie/bwa/spa/heating-mode/$name Heating Mode
homie/bwa/spa/heating-mode/$datatype enum
homie/bwa/spa/heating-mode/$format ready,rest,ready_in_rest
homie/bwa/spa/heating-mode/$settable true
homie/bwa/spa/temperature-scale fahrenheit
homie/bwa/spa/temperature-scale/$name Temperature Scale
homie/bwa/spa/temperature-scale/$datatype enum
homie/bwa/spa/temperature-scale/$format fahrenheit,celsius
homie/bwa/spa/temperature-scale/$settable true
homie/bwa/spa/twenty-four-hour-time false
homie/bwa/spa/twenty-four-hour-time/$name 24 Hour Time
homie/bwa/spa/twenty-four-hour-time/$datatype boolean
homie/bwa/spa/twenty-four-hour-time/$settable true
homie/bwa/spa/temperature-range high
homie/bwa/spa/temperature-range/$name Temperature Range
homie/bwa/spa/temperature-range/$datatype enum
homie/bwa/spa/temperature-range/$format high,low
homie/bwa/spa/temperature-range/$settable true
homie/bwa/spa/current-temperature 99
homie/bwa/spa/current-temperature/$name Current Water Temperature
homie/bwa/spa/current-temperature/$datatype float
homie/bwa/spa/current-temperature/$format 32:108
homie/bwa/spa/current-temperature/$unit °F
homie/bwa/spa/target-temperature 99
homie/bwa/spa/target-temperature/$name Target Water Temperature
homie/bwa/spa/target-temperature/$datatype float
homie/bwa/spa/target-temperature/$format 50:106
homie/bwa/spa/target-temperature/$settable true
homie/bwa/spa/target-temperature/$unit °F
homie/bwa/spa/circulation-pump true
homie/bwa/spa/circulation-pump/$name Circulation Pump Running
homie/bwa/spa/circulation-pump/$datatype boolean
homie/bwa/filter-cycle1/$name Filter Cycle 1
homie/bwa/filter-cycle1/$type Filter Cycle
homie/bwa/filter-cycle1/$properties running,start-hour,start-minute,duration
homie/bwa/filter-cycle1/running false
homie/bwa/filter-cycle1/running/$name Running
homie/bwa/filter-cycle1/running/$datatype boolean
homie/bwa/filter-cycle1/start-hour 13
homie/bwa/filter-cycle1/start-hour/$name Start Hour
homie/bwa/filter-cycle1/start-hour/$datatype integer
homie/bwa/filter-cycle1/start-hour/$format 0:23
homie/bwa/filter-cycle1/start-hour/$settable true
homie/bwa/filter-cycle1/start-hour/$unit hours
homie/bwa/filter-cycle1/start-minute 0
homie/bwa/filter-cycle1/start-minute/$name Start Minute
homie/bwa/filter-cycle1/start-minute/$datatype integer
homie/bwa/filter-cycle1/start-minute/$format 0:59
homie/bwa/filter-cycle1/start-minute/$settable true
homie/bwa/filter-cycle1/start-minute/$unit minutes
homie/bwa/filter-cycle1/duration 180
homie/bwa/filter-cycle1/duration/$name Duration
homie/bwa/filter-cycle1/duration/$datatype integer
homie/bwa/filter-cycle1/duration/$format 0:1439
homie/bwa/filter-cycle1/duration/$settable true
homie/bwa/filter-cycle1/duration/$unit minutes
homie/bwa/filter-cycle2/$name Filter Cycle 2
homie/bwa/filter-cycle2/$type Filter Cycle
homie/bwa/filter-cycle2/$properties running,start-hour,start-minute,duration,enabled
homie/bwa/filter-cycle2/running false
homie/bwa/filter-cycle2/running/$name Running
homie/bwa/filter-cycle2/running/$datatype boolean
homie/bwa/filter-cycle2/start-hour 8
homie/bwa/filter-cycle2/start-hour/$name Start Hour
homie/bwa/filter-cycle2/start-hour/$datatype integer
homie/bwa/filter-cycle2/start-hour/$format 0:23
homie/bwa/filter-cycle2/start-hour/$settable true
homie/bwa/filter-cycle2/start-hour/$unit hours
homie/bwa/filter-cycle2/start-minute 0
homie/bwa/filter-cycle2/start-minute/$name Start Minute
homie/bwa/filter-cycle2/start-minute/$datatype integer
homie/bwa/filter-cycle2/start-minute/$format 0:59
homie/bwa/filter-cycle2/start-minute/$settable true
homie/bwa/filter-cycle2/start-minute/$unit minutes
homie/bwa/filter-cycle2/duration 15
homie/bwa/filter-cycle2/duration/$name Duration
homie/bwa/filter-cycle2/duration/$datatype integer
homie/bwa/filter-cycle2/duration/$format 0:1439
homie/bwa/filter-cycle2/duration/$settable true
homie/bwa/filter-cycle2/duration/$unit minutes
homie/bwa/filter-cycle2/enabled/$name Enabled
homie/bwa/filter-cycle2/enabled/$datatype boolean
homie/bwa/filter-cycle2/enabled/$settable true
homie/bwa/spa/pump1/set false
homie/bwa/$state lost
homie/bwa/$state init
homie/bwa/$state ready
homie/bwa/spa/pump1/set false
homie/bwa/$state lost
homie/bwa/$state init
homie/bwa/$state ready
```

BWA Logs
The same 2 lines repeat forever when controlling the pumps.
```
D,  read: 7e 07 10 bf 11 00 00 3e 7e
D, from spa: #<BWA::Messages::ToggleItem 0>
```

jshank, 2022-01-08:
I can turn off pumps in BWA 1.3.0 (ghcr.io/jshank/bwalink:2021.12) but nothing later (bwalink:2021.12.1 or bwalink:2021.12.2). Sorry that I never tested turning on and off pumps in the last few versions.

jshank, 2022-01-08:
Interestingly enough, `toggle` still works properly.

```
homie/bwa/spa/pump1/set toggle
homie/bwa/spa/pump1 true
homie/bwa/spa/pump1/set toggle
homie/bwa/spa/pump1 false
```

ccutrer, 2022-01-17:
I'm sure it's something simple. can you set Thread.abort_on_exception to true temporarily, and hopefully it spits out an exception.

jshank, 2022-01-18:
No change with `Thread.abort_on_exception = true` in `bwa_mqtt_bridge` line 26

The log is just filled with the ToggleItem 0 messages after the initialization 

```
D,  read: 7e 07 10 bf 11 00 00 3e 7e
bwalink    | D, from spa: #<BWA::Messages::ToggleItem 0>
```

ccutrer, 2022-01-18:
man, I got nothing. the configuration is clearly returning that the pump has 1 available speed (it must be 1 to publish it as a boolean to MQTT). the current pump speed is showing as true, so it's _not_ zero. the code assumes it's going to be 1. so that's something that might be off. I'm also not sure why you're _receiving_ toggle item messages, especially with item id 0, which the code doesn't recognize as any particular item. it feels like somehow the math in Client#set_pump is wrong causing it to send the toggle item message too many times, so you might want to add some additional logging there to see how many times it tries to toggle it.

jshank, 2022-01-18:
I think I figured it out, but help me with my assumptions. The only thing you can do to a pump is call toggle (analog to pushing the button). In the code, you are trying to determine the number of times to toggle to get the pump into the desired state. It looks like the math pushes the button once for turning it on works but you have it pushed zero times for off which does nothing. I added some debugging code with the following results:

BWA Debug

```
W, Pump 0 speed detected as 1
W, Pump 1 speed detected as 1
W, Pump 2 speed detected as 1
W, Balboa MQTT Bridge running (version 2.0.6)
```

Toggled Pump 1 ON
```
W, Setting pump 0 to true
W, set_pump: Pump 0 setting to true
W, set_pump: Not returning
W, set_pump: false check - Changing desired to true
W, set_pump: min check - Changing desired to 1
W, set_pump: times calc - (desired: 1 - status.pumps[index]: 0) % (configuration.pumps[index]: 1 + 1) = 1
W, set_pump: Executing toggle times set to 1
W, set_pump: Calling toggle_pump on pump 0
```

Toggled Pump 1 off
```
W, Setting pump 0 to false
W, set_pump: Pump 0 setting to false
W, set_pump: Not returning
W, set_pump: false check - Changing desired to 0
W, set_pump: min check - Changing desired to 0
W, set_pump: times calc - (desired: 0 - status.pumps[index]: 2) % (configuration.pumps[index]: 1 + 1) = 0
W, set_pump: Executing toggle times set to 0
```

Notice the lack of a  `Calling toggle_pump on pump X`

client.rb debug changes
```ruby
    def set_pump(index, desired)
      BWA.logger.warn "set_pump: Pump #{index} setting to #{desired}"
      return unless status && configuration
      BWA.logger.warn "set_pump: Not returning"
      desired = 0 if desired == false
      BWA.logger.warn "set_pump: false check - Changing desired to #{desired}"
      desired = configuration.pumps[index] if desired == true
      desired = [desired, configuration.pumps[index]].min
      BWA.logger.warn "set_pump: min check - Changing desired to #{desired}"
      times = (desired - status.pumps[index]) % (configuration.pumps[index] + 1)
      BWA.logger.warn "set_pump: times calc - (desired: #{desired} - status.pumps[index]: #{status.pumps[index]}) % (configuration.pumps[index]: #{configuration.pumps[index]} + 1) = #{(desired - status.pumps[index]) % (configuration.pumps[index] + 1)}"
      BWA.logger.warn "set_pump: Executing toggle times set to #{times}"
      times.times do
        BWA.logger.warn "set_pump: Calling toggle_pump on pump #{index}"
        toggle_pump(index)
        sleep(0.1)
      end
    end
```

ccutrer, 2022-01-18:
`W, set_pump: times calc - (desired: 0 - status.pumps[index]: ***2***)```

there it is. my assumption was that a single speed pump would return speed 1 when it's on, but it's returning speed 2, so the math is wrong. I'll get a fix up.

ccutrer, 2022-01-18:
🤞  2.0.7 fixes it for you

closed 2022-01-18 by ccutrer

jshank, 2022-01-18:
Confirmed working, thanks @ccutrer!

commit 9dc7283b (2022-01-18): fix turning off single speed pumps

fixes #53

they report their current speed as 2, even though they only have 1
speed. so limit current speed to max speed, so that the math works
out for both single and dual speed pumps