ccutrer/balboa_worldwide_app #55: Docs Error - Pi + RS485

netmindz, 2022-01-15:
The docs state "It is possible to directly connect to the GPIO on a Raspberry Pi" but fro what I can see, none of the models of the Pi support RS485 directly, only with the addition of UART to RS485 adapter shield

ccutrer, 2022-01-17:
I apologize. I should have said something more along the lines of "using the GPIO pins connected to a RS-485 transceiver, rather than a USB adapter".  An example transceiver is the MAX485. But helping someone connect such a device and set it up as a serial device in Linux is beyond the scope of this project.

closed 2022-01-17 by ccutrer

netmindz, 2022-01-17:
The regular max485 can't be used as you need to toggle high/low to control when in transmit or receive, apparently the newer adapters like ARCELI TTL To RS485 Adapter 485 can handle that, but yet to test mine

netmindz, 2022-01-17:
Example of how you need to toggle the pin when using max485 https://github.com/cribskip/esp8266_spa/blob/master/src/balboa_helper.ino

ccutrer, 2022-01-17:
ah yes, good point. https://www.amazon.com/gp/product/B07B667STP is what I have, and while I haven't used it either with my spa or a pi, I have used it with RS485 and an ESP to build a serial bridge, and I didn't need to mess with additional pins. you could probably get a MAX485 to work if you wrap it with some software that will set the read/write pin for you so that it presents either as an OS-level serial device, or is accessible over the network.

netmindz, 2022-01-18:
Readme still says MAX485 rather than the MAX13487 used in that

commit 1d7f0f8e (2022-01-17): update docs to clarify "direct" GPIO connection

closes #55