ccutrer/balboa_worldwide_app #57: 8-pin pinout?

VIDGuide, 2022-01-30:
So there is heaps of (amazing!) information here, but I seem to be coming up a little short. 

I've got a GS-510DZ. It has a J1 and J2 connector, both RJ45. J1 goes to the control panel. 

There are markings on the pcb for J1A and J2A, which is where the Molex connectors seem to be on other models in the family. To the point it seems that it's the same thing, just different connectors. 

The fact they aren't populated on mine is fine, I'm happy enough to solder in a connector, or if the RJ45 socket *is* the same thing, then to use that instead. 

PCB traces for the 2 "A" connectors appear to be connected in series, suggesting rs485. 

In fact I can see the J1/J1A has extra connectors on the top 2 pins, which fits with the manual warnings that back-lit control panels will only work on J1, I'm guessing it's an extra power supply for the leds. 

[image]

So in short, I believe there is rs485 here, but I'm confused as to which pins it will be on in the 8-pin connector. 

I'll assume not pins 1-2, due to the extra power and non-connection to J2. Looks like pin 8 is also connected, but pin 7 is not connected on J1, connected elsewhere on J2. 

So I'm going to guess the RS485 is pins 3,4,5,6 out of the 8, with 1-2 for extra power and 7-8 unknown. 

What's the best way to safely test for rs485 here? I don't want to kill it :)

bggardner, 2022-01-30:
This issue is covered in #14.  We are fairly certain the 8-pin connections do not use RS-485, but another (probably proprietary) serial protocol, which we have not yet documented.

closed 2022-01-30 by bggardner

netmindz, 2022-02-07:
@VIDGuide Not sure how similar the GS is to the GL range, but you might want to follow the work here https://github.com/netmindz/balboa_GL_ML_spa_control as at controller of similar age

VIDGuide, 2022-02-08:
Thanks @netmindz - It does look similar. I'll check it out. I've actually got access to an oscilloscope, so might have to use that to try to track out it's pinout too.

netmindz, 2022-03-04:
> Thanks @netmindz - It does look similar. I'll check it out. I've actually got access to an oscilloscope, so might have to use that to try to track out it's pinout too.

Any luck?

netmindz, 2022-06-27:
@VIDGuide did you have a look at the project?

VIDGuide, 2022-06-28:
I certainly did, but I haven't had a chance to return to the work at all on my side. It's winter for me where I am now, so the pool and things around it as a low priority. I'm sure as we swing back around to summer and I wake the pool and spa up, I'll dig back into this :)

ccutrer, 2022-06-28:
Totally agree. Winter is the _best_ time to use a tub :)