ccutrer/balboa_worldwide_app #58: Unknown Messages / Partial Communication

HyperActiveJ, 2022-03-04:
Hello,
I need some help decoding w messages from my 2021 Sundance 780 series spa. I recognize this is not a supported main board by the balboa worldwide app but I'm not sure where else to ask this.

 I successfully connected a RS485<>Ethernet converter as described here: https://github.com/jshank/bwalink. I first attempted to use the this app as well as pybalboa unmodified with out success. I turned to manually adapting pybalboa to print raw data to try to identify the issue

I demonstrated I think I have my hardware working as expected by succeeding at implementing the channel assignment sequence:
1.	Wait for New Client Clear to Send (Confirmed at 1hz for my board)
2.	Sent Channel Assignment Request
3.	Received Channel Assignment Response (Channel assigned started at 0x20 and incrementing by 1 each time)
4.	Sent Channel Assignment Ack
5.	Main board began to send Clear To Send (CTS) Messages for the newly assigned channel
6.	Returned Nothing to Send response and see main board moving to next channel upon receipt. 
7.     Repeated steps above and received channel assignments incremented by 1 each time

With some confidence I then tired to send the following settings request messages (Type 0x22) after a CTS message in-place of a Nothing To Send message, with either codes 0, or 4 (Configuration, Information). However I observed no response to these messages.  I also tried hard coding channel 0x0A (Wifi) without success.

Further, I am not receiving the known Status Update (0x13) message, nor was I able to capture the control panel sending any known message types by pressing controls. Instead I only receive a 3 different types of message at > 1hz rates.

Channel : Message Type: Number of Fields
0x10 : 0xCC : 2
0xFF : 0xCA : 29
0xFF : 0xC4 : 33

Example messages:
Ch: 0x10 Type: 0xCC Len 7 - Data ['B1', '51']
Ch: 0xFF Type: 0xC4 Len 38 - Data ['35', '21', '31', '30', '3F', '3F', 'A6', 'C6', '3E', '3A', '5C', '3A', '07', '26', 'C5', '26', '47', '22', '20', '36', '2F', '2E', '01', '2C', 'B0', '2A', '29', '7C', '72', '16', '15', '14', '13']
Ch: 0x10 Type: 0xCC Len 7 - Data ['A4', '44']
Ch: 0xFF Type: 0xC4 Len 38 - Data ['C7', 'D3', 'C3', 'C2', 'CD', 'CD', '54', '34', 'CC', 'C8', 'AE', 'C8', 'F5', 'D4', '37', 'D4', 'B5', 'D0', 'D2', 'C4', 'DD', 'DC', 'F3', 'DE', '42', 'D8', 'DB', '8E', '80', 'E4', 'E7', 'E6', 'E1']
Ch: 0xFF Type: 0xCA Len 34 - Data ['C3', '7F', '83', '82', '8D', '8C', '8F', 'B2', '89', '88', '8B', '75', '95', '94', '97', '96', '91', '90', '93', '92', '9D', '9C', '9F', '9E', '99', '98', '9B', '9A', 'A5']
Ch: 0x10 Type: 0xCC Len 7 - Data ['B2', '52']
Ch: 0xFF Type: 0xC4 Len 38 - Data ['9F', '8B', '9B', '9A', '95', '95', '0C', '6C', '94', '90', 'F6', '90', 'AD', '8C', '6F', '8C', 'ED', '88', '8A', '9C', '85', '84', 'AB', '86', '1A', '80', '83', 'D6', 'D8', 'BC', 'BF', 'BE', 'B9']
Ch: 0xFF Type: 0xC4 Len 38 - Data ['39', '2D', '3D', '3C', '33', '33', 'AA', 'CA', '32', '36', '50', '36', '0B', '2A', 'C9', '2A', '4B', '2E', '2C', '3A', '23', '22', '0D', '20', 'BC', '26', '25', '70', '7E', '1A', '19', '18', '1F']
Ch: 0x10 Type: 0xCC Len 7 - Data ['1E', 'FE']
Ch: 0xFF Type: 0xC4 Len 38 - Data ['03', '17', '07', '06', '09', '09', '90', 'F0', '08', '0C', '6A', '0C', '31', '10', 'F3', '10', '71', '14', '16', '00', '19', '18', '37', '1A', '86', '1C', '1F', '4A', '44', '20', '23', '22', '25']
Ch: 0x10 Type: 0xCC Len 7 - Data ['DF', '3F']

My observations are:
1.	The lower nibble of message type 0xCC is always identical for both data bytes.
2.	For both message types 0xCA and 0xC4: For a given first data byte, most messages received will be exactly identical. For example if the first data byte of message type 0xC4 is 0x39, then for most messages the rest of the data will be '2D', '3D', '3C', '33', '33', 'AA', 'CA', '32', '36', '50', '36', '0B', '2A', 'C9', '2A', '4B', '2E', '2C', '3A', '23', '22', '0D', '20', 'BC', '26', '25', '70', '7E', '1A', '19', '18', '1F'
3.     If there are differences in messages for a given message type and first data byte, there are at most 5 bytes different, typically with only one or two bits within any given byte being alerted.
4.	When Message of types CA or C4 contain byte values different from first observed, the difference tends to be the same for several messages before returning to its original values, with a couple of exceptions that may be indicative of the values representing a longer term trend such as possibly representing temperature.
5.	There appears to be no pattern or correlation between the message types, or from message to message that I see so far (ex, no incrementing counters, no consistency in order).

So,
I’m hoping someone might be able to point me in the right direction to try to narrow in further. Any ideas?

bggardner, 2022-03-04:
Kudos for getting as far as you have.  My guess is that these new message types may be new "Status Update" messages, but because their field values are so different between messages that may not be the case.  You may need to post a timestamped dump/log of a message stream to make more sense of these.  It would be even better if the log was notated to say when "obvious" telemetry like temperatures change on the display/control panel.

FYI, you may want to switch to a RS-485 to USB adapter, as any IP-based system may introduce delays and/or drop packets.  The Ethernet converter is probably fine, but switching to a lower-level serial device may help with troubleshooting.

I think investigating the issue about not being able to detect control panel messages is critical.  Does it use the same kind of connector?  Are the control panel connector signal "bussed" with the RS-485<>Ethernet connector signals (does pin 1 connect to pin 1, etc.)?  Maybe post some pictures.  What kind of traffic is there if the control panel is unplugged?  Also try logging the bus during power-on, both the the control panel plugged in and disconnected.

HyperActiveJ, 2022-03-04:
Ordered usb 485 converter.

3 of same connector outiled with a silkscren box on the board. Diagram on box indicates smarttub module can plug into same connector area. Moved panel cable over to the connector i had been using and panel continued to work as expected so it doesnt seem to care where its attached at least. Will buzz it out and confrim board really has the connectors bussed and switch to a y cable if nessary. Agreed not seening panel messages is a problem for decoding. I see cts messages for other channels but no info from them. Will add photos.

Is there a perfered log format? Hex dump with timestamps?

bggardner, 2022-03-04:
Hex dump with timestamps is good, with manual notations of known events.  See the wiki for some similar examples.

I wonder if the new panels use wireless comm, and just get power via the wires?  Can you tell if the control panel uses all 4 wires?  Are there metal contacts in all four connector positions of the cable?  Is there an FCC ID anywhere on the panel or main board?  This would indicate wireless comm.

bggardner, 2022-03-07:
Summary/Notes:
- Message Type `0x00` (New Client Clear to Send) is broadcast at ~10 Hz (as expected, probably reduces to 1 Hz after a few more seconds).
- Message Type `0x06` (Clear to Send) is sent somewhat sporadically, averaging maybe 60Hz (expecting 30Hz).  The control panel responds to the first four (after channel assignment) with Message Type `0x17` and payload `0x00`.  The subsequent responses are of Message Type `0xCC` with payload consisting of two bytes:
    - Least-significant nibbles are always identical (though not the same value between messages)
    - Most-significant nibbles always appear in pairs: `0/E`, `1/F`, `2/C`, `3/D`, `4/A`, `5/B`, `6/8`, `7/9`
        - Nibbles in the pair always have the same parity (even/odd)
        - Even nibble pairs sum to 14
        - Odd nibble pairs sum to 16
- Message Type `0xC4` is broadcast at ~3 Hz (similar to Status Updates)
    - Example payloads (in order of appearance), with first row denoting byte number:
        - `00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20`
        - `90 9C 94 95 9A 9B 0D 63 9B 9F F8 9F 92 83 B0 82 E3 87 85 93 8A 8B 99 89 1B 8F 8E C3 D6 B3 B0 B1 B6`
        - `51 5D 55 54 5B 5A CC A2 5A 5E 39 5E 53 42 71 43 22 46 44 52 4B 4A 58 48 DA 4E 4F 02 17 72 71 70 77`
        - `72 7E 76 77 78 79 EF 81 79 7D 1A 7D 70 61 52 60 01 65 67 71 68 69 7B 6B F9 6D 6C 21 34 51 52 53 54`
        - `F6 FA F2 F3 FC FD 6B 05 FD F9 9E F9 F4 E5 D6 E4 85 E1 E3 F5 EC ED FF EF 7D E9 E8 A5 B0 D5 D6 D7 D0`
        - `9E 92 9A 9B 94 95 03 6D 95 91 F6 91 9C 8D BE 8C ED 89 8B 9D 84 85 97 87 15 81 80 CD D8 BD BE BF B8`
        - `CA C6 CE CF C0 C1 57 39 C1 C5 A2 C5 C8 D9 EA D8 B9 DD DF C9 D0 D1 C3 D3 41 D5 D4 99 8C E9 EA EB EC`
        - `7D 71 79 78 77 76 E0 8E 76 72 15 72 7F 6E 5D 6F 0E 6A 68 7E 67 66 74 64 F6 62 62 2E 3B 5E 5D 5C 5B`
        - `AA A6 AE AF A0 A1 37 59 A1 A5 C2 A5 A8 B9 8A B8 D9 BD BF A9 B0 B1 A3 B3 21 B5 B5 F9 EC 89 8A 8B 8C`
        - `A7 AB A3 A2 AD AC 3A 54 AC A8 CF A8 A5 B4 87 B5 D4 B0 B2 A4 BD BC AE BE 2C B8 B8 F4 E1 84 87 86 81`
        - `10 1C 14 15 1A 1B 8D E3 1B 1F 78 1F 12 03 30 02 63 07 05 13 0A 0B 19 09 9B 0F 0F 43 56 33 30 31 36`
        - `B0 BC B4 B5 BA BB 2D 43 BB BF D8 BF B2 A3 90 A2 C3 A7 A5 B3 AA AB B9 A9 3B AF AF E3 F6 93 90 91 96`
        - `41 4D 45 44 4B 4A DC B2 4A 4E 29 4E 43 52 61 53 32 56 54 42 5B 5A 48 58 CA 5E 5E 12 07 62 61 60 67`
        - `F9 F5 FD FC F3 F2 64 0A F2 F6 91 F6 FB EA D9 EB 8A EE EC FA E3 E2 F0 E0 72 E6 E6 AA BF DA D9 D8 DF`
        - `A2 AE A6 A7 A8 A9 3F 51 A9 AD CA AD A0 B1 82 B0 D1 B5 B7 A1 B8 B9 AB BB 29 BD BA F1 E4 81 82 83 84`
        - `47 4B 43 42 4D 4C DA B4 4C 48 2F 48 45 54 67 55 34 50 52 44 5D 5C 4E 5E CC 58 5F 14 01 64 67 66 61`
    - Payload bytes `0x00`-`0x05`, `0x08`-`0x09`, `0x0B`-`0x0C`, `0x13`, and `0x16` have the same most-significant nibble
    - Payload bytes `0x0D`, `0x0F`, `0x11`, `0x12`, `0x14`-`0x15`, `0x17`, `0x19`, and `0x1A` have the same most-significant nibble, which is one bit/integer higher than the most-significant nibble of byte `0x00`
- Message Type `0xCA` is broadcast at ~1 Hz.
    - Example payloads (in order of appearance), with first row denoting byte number:
        - `00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C`
        - `F3 4F B3 B2 BD BC BF 82 B9 B8 BB 45 A5 A4 A7 A6 A1 A0 A3 A2 AD AC AF AE A9 A8 AB AA 95`
        - `EB 57 AB AA A5 A4 A7 9A A1 A0 A3 5D BD BC BF BE B9 B8 BB BA B5 B4 B7 B6 B1 B0 B3 B2 8D`
        - `54 E8 14 15 1A 1B 18 25 1E 1F 1C E2 02 03 00 01 06 07 04 05 0A 0B 08 09 0E 0F 0C 0D 32`
        - `DB 67 9B 9A 95 94 97 AA 91 90 93 6D 8D 8C 8F 8E 89 88 8B 8A 85 84 87 86 81 80 83 82 BD`
        - `54 E8 14 15 1A 1B 18 25 1E 1F 1C E2 02 03 00 01 06 07 04 05 0A 0B 08 09 0E 0F 0C 0D 32`
    - Payload bytes `0x02`-`0x06` and `0x08`-`0x0A` have the same most-significant nibble
    - Payload bytes `0x0C`-`0x1B` have the same most-significant nibble (but not the same as in byte `0x02`)

I think the next step is to only log Message Types `0xC4` and `0xCA` (along with timestamps) and try to match the timestamps with events (temperature change, control panel command, etc.).  You may also want to just log `0xCC` messages (or any non-CTS messages on channel `0x10`) during control panel button presses.

HyperActiveJ, 2022-03-07:
Very interesting observation there about the nibbles...
I've attached a full startup log.
Temperature was 99F, the system went thru  its normal startup cycle of pump 1 then pump 2 then on to just circulating pump.
I attempted to get some data while pressing buttons in a new log file, but it looks like I lost bytes and don't see even partial messages associated with the button presses. Need to figure out if I've got a hardware issue (eg noise)  from changing my test cables last night, if my app hickuped, or if the button presses is inducing the issue directly.

[202203061110StartupNoButtons.txt](https://github.com/ccutrer/balboa_worldwide_app/files/8201276/202203061110StartupNoButtons.txt)

bggardner, 2022-03-15:
Still can't make sense of the data.  You need to capture the data just before and after a change (current temperature or button press).  Try to limit the time surrounding the change event to a minimum (a few seconds).  The idea is to compare the data before and after the change to try and detect what byte(s) is/are affected.  Example:

1. Note set temperature and start recording.
2. Wait a few seconds.
3. Increase set temperature one degree.
4. Wait a few seconds.
5. Stop recording and save as something like "SetTemp_99to100.txt"

Repeat this procedure for other button pushes, like decreasing set temperature, turning on/off lights and pumps, naming the file appropriately.

Also, is [this](https://shop.aqua-tech.ca/products/sundance-spas-control-panel-lcd-p-n-6600-385) what your control panel looks like?  If not, can you post a picture of or link to it?

HyperActiveJ, 2022-03-15:
> While unsupported, certainly looks like the messages are more similar than the older GL/EL Balboa series

Thats an interesting theory, how so?

HyperActiveJ, 2022-03-15:
> Still can't make sense of the data. You need to capture the data just before and after a change (current temperature or button press). Try to limit the time surrounding the change event to a minimum [...]

That is my control panel!
Sorry I've been on travel and haven't been able to make another capture, hopefully next weekend. I have a new script so I can press a keyboard key to record what I pressed on the panel and get a corresponding  log to it. appreciate your look at it!

HyperActiveJ, 2022-03-21:
Alright, here's the captures, still not making sense of it :(.

Order was:
Pump 1 on
Pump 1 off
Pump 2 on
Pump 2 off
Temp to 101
Temp to 100

I started the log about one send before pressing, then stopped log a second or so after the change registered or pump had turned on/off.

[TempSetTo101.txt](https://github.com/ccutrer/balboa_worldwide_app/files/8319261/TempSetTo101.txt)
[Pump2Off.txt](https://github.com/ccutrer/balboa_worldwide_app/files/8319262/Pump2Off.txt)
[Pump2On.txt](https://github.com/ccutrer/balboa_worldwide_app/files/8319263/Pump2On.txt)
[Pump1Off.txt](https://github.com/ccutrer/balboa_worldwide_app/files/8319264/Pump1Off.txt)
[Pump1On.txt](https://github.com/ccutrer/balboa_worldwide_app/files/8319265/Pump1On.txt)
[TempSetTo100.txt](https://github.com/ccutrer/balboa_worldwide_app/files/8319266/TempSetTo100.txt)

HyperActiveJ, 2022-04-04:
Performing bitwise xor to compare consecutive bytes in the C4 messages produces a result which is identical for all C4 messages sent while the hotub is in a steady state. Further I was able to find a pattern in changes to the result of this function which tracks with changes in pump state. IE given the latest  C4 message I can determine what the bitwise xor results will change to in response to changes in any given pump state. Or in otherwords I think I'm reading pump share reliably. The odd thig is that multiple bytes of the result change. So I don't quite see the next level of what this means in terms of decrypting the message, I'm hoping that this is indicating a standard Cypher of some sort that can be generalized. Still working it.

HyperActiveJ, 2022-04-04:
The bitwise xor of consecutive bytes also is identical for all CA messages in steady state.

HyperActiveJ, 2022-04-09:
I'm able now to decode from the 0xC4  message the hot tubs currently set Hours, Minutes, Set Setpoint seemingly reliably from startup with no prior knowledge of sate, and Temperature when pumps are not on or if the pump state is known (not sure why yet),

Python Source.
data = datafieldsint(fields)
currentMsg= []
for i in range(0,len(data)-1,2):
          c = data[i]^data[i+1]
          currentMsg.append(c)            
if (currentMsg != lastMsg):
           print("{} H{} M{} ST{} T{} x{}".format(fields[0], lst[0]^7, lst[11]^1, lst[8]^1, lst[14], lst))
           lastMsg= currentMsg 

Example Output while pressing the temp down to 65 then back up again with a current temp of 99.

 (H= hours, M = Minute, ST = Set point, T= Temperature)

16:24:58.447 H16 M24 ST70 T99 x[23, 1, 36, 142, 4, 96, 37, 69, 71, 22, 1, 25, 116, 46, 99, 1]
16:24:59.317 H16 M24 ST69 T99 x[23, 1, 36, 142, 4, 96, 37, 69, 68, 22, 1, 25, 116, 46, 99, 1]
16:24:59.902 H16 M24 ST68 T99 x[23, 1, 36, 142, 4, 96, 37, 69, 69, 22, 1, 25, 116, 46, 99, 1]
16:25:00.534 H16 M24 ST67 T99 x[23, 1, 36, 142, 4, 96, 37, 69, 66, 22, 1, 25, 116, 46, 99, 1]
16:25:01.130 H16 M25 ST66 T99 x[23, 1, 36, 142, 4, 96, 37, 69, 67, 22, 1, 24, 116, 46, 99, 1]
16:25:01.977 H16 M25 ST65 T99 x[23, 1, 36, 142, 4, 96, 37, 69, 64, 22, 1, 24, 116, 46, 99, 1]
16:25:07.452 H16 M25 ST66 T99 x[23, 1, 36, 142, 4, 96, 37, 69, 67, 22, 1, 24, 116, 46, 99, 1]
16:25:07.997 H16 M25 ST67 T99 x[23, 1, 36, 142, 4, 96, 37, 69, 66, 22, 1, 24, 116, 46, 99, 1]
16:25:08.576 H16 M25 ST68 T99 x[23, 1, 36, 142, 4, 96, 37, 69, 69, 22, 1, 24, 116, 46, 99, 1]
16:25:09.211 H16 M25 ST69 T99 x[23, 1, 36, 142, 4, 96, 37, 69, 68, 22, 1, 24, 116, 46, 99, 1]

HyperActiveJ, 2022-04-11:
Narrowing in further, got more of the status decoded but a few aren't fields reliable yet.

Made a break thur on the  CC message - its just corresponds to Key pressed!
field1 xor^ field 2 decodes as the following:

225 - Nothing Pressed
238 Clear ray button
229 Pump 1 Button
228 Pump 2 Button
227 Temp Down
224 Temp Up
255 Menu

Next i need to try sending these to see if I can induce a response..

HyperActiveJ, 2022-04-14:
I now have 'the basics' working - Reading and writing Temperatures, pumps and lights. No more advanced features like filter configuring filter cycles.
I will submit a updates to the wiki.
I am working toward maturing a fork of pybalboa which implements it cleanly.

closed 2025-05-12 by ccutrer