ccutrer/balboa_worldwide_app #61: Feature request: show and set spa time

pixiandreas, 2022-03-27:
When i check the official phone app i can see that it possible to set and see the actual time.
Is this something that can be added ?

flechaig, 2022-03-28:
Hi,
THis is what I did:

    String sHour = payload.substring(0,2);
     String sMinute = payload.substring(3);
    hourSet = sHour.toInt();
    // The highest bit enables 24-hour format: add 0x80 (=128) to the hour in the 24 format
    // Send 0x94 for 20 (vingt heures)
    hourSet = hourSet + 128;
    minuteSet = sMinute.toInt();
    //mqtt.publish("Spa/node/debug", String(hourSet).c_str());
    send = 0xee;

then

          uint8_t h = hourSet;
          uint8_t m = minuteSet;
          Q_out.push(id);
          Q_out.push(0xBF);
          Q_out.push(0x21);
          Q_out.push(h);
          Q_out.push(m);

and get current time:

          // 8:Flag Byte 3 Hour & 9:Flag Byte 4 Minute => Time
          if (Q_in[8] < 10) s = "0"; else s = "";
          SpaState.hour = Q_in[8];
          s += String(Q_in[8]) + ":";
          if (Q_in[9] < 10) s += "0";
          s += String(Q_in[9]);
          SpaState.minutes = Q_in[9];
          mqtt.publish("Spa/time/state", s.c_str());

Hope it helps!

closed 2022-03-28 by ccutrer

pixiandreas, 2022-03-28:
I checked the time on the spa and it's in sync, sorry i missed to check it before i create the issue.
I think is a good idea to add that the time is automatically synchronized in the documentation ?