ccutrer/balboa_worldwide_app #63: Toggle item 29 (0x1D)

fastfourier666, 2022-06-13:
Hi and thanks for this software. It works straight away with my controller (MQBP20UX) and I can control it with Home Assistant.

I saw a toggle item which is not in the wiki: number 29 (0x1D) which is generated by pressing the "soak" button on my control panel. This button turns all the pumps off.

I'm not sure how widely supported it is, but it might make for a better method of turning the pump off, rather than cycling the pump mode back to 0. Currently when I select 0 for the pump speed, sometimes it misses a toggle and is stuck on 2.

I will try to experiment by adding this command to the code, but I am very new to ruby.

thanks!

closed 2022-06-13 by ccutrer

fastfourier666, 2022-06-14:
Wow thank you! I'll try this out.

bggardner, 2022-06-14:
FYI, `0x1D` has no effect for me, so it is not supported by all controllers.