ccutrer/balboa_worldwide_app #65: Controls go "Unavailable" (Home Assistant)

fastfourier666, 2022-07-16:
Hi C,

Sorry, I seem to be the only one banging on about stuff in your issues recently! I am still using this and it's been great for managing power usage on our tub.

I am using this on an Ubuntu 20.04 proxmox VM. It's a fresh install apart from this library (installed with instructions in the README) and I'm using ruby 2.7.0p0. It talks to HA/Mosquitto running on another VM. The tub controller is MQBP20UX V2.1.1, connected with USB -> RS485 converter via USB passthrough from the host machine. All works fine so far. 

At around 8:49PM every night the controls in HA become "Unavailable" (greyed out).  The clock and timezone is set correctly. For the last few nights it's happened at 20:39:23, 20:39:23 again, 20:39:35, 20:49:47 and 20:49:41 this evening.

I poked around for a few days and found:

- It does not appear to be related to the timed settings of the tub, like filter cycle and duration.
- HA and physical controls work fine just before it happens (and all other times in the day).
- Looking at the MQTT logs at the witching hour, the ```homie/bwa/$state``` topic gets set to ```init``` then ```ready```, then 3 seconds later ```lost```.
- I have the systemd file set to DEBUG with LOG_VERBOSITY=1 and nothing unusual is listed in ```journalctl -u bwa_mqtt_bridge.service```.
- Nothing unusual in syslog either.
- Nothing unusual is happening in the HA Logbook at the time, apart from all the bwa controls suddenly getting listed as unavailable.
- After the HA controls stop working, I can look at the homie topics using MQTT Explorer and the physical tub controls still update them.
- If I manually set ```/homie/bwa/$state``` back to ```ready```, I can once again control the tub from HA.

Any idea what this could be, or the best way to proceed to debugging it? I know next to nothing about Ruby unfortunately. Right now my "solution" is to monitor one of the bwa entities in Home Assistant and fire an ssh command at the balboa VM to restart the service. It works, but I'm more curious than anything about what could be causing this...

ccutrer, 2022-07-18:
Hmm, that's a bit unusual. The only way `$state` can go to `lost` is by the MQTT server seeing the connection close, and applying the Last Will and Testament message. But at that point obviously no more messages can be published. The client will automatically reconnect, but when it does so it should be automatically re-publishing `ready`. Which MQTT server are you using?

fastfourier666, 2022-07-18:
I'm using Mosquitto 6.1.2 which is installed as a Home Assistant add-on.

Could the three-second gap between the messages be related to the ```RestartSec=3s``` in the service file?

fastfourier666, 2022-07-20:
Well, I think this was probably something wonky with the virtual machine.

I had this problem about three days in a row, and each time it took me anywhere between a few minutes and a few hours to realise it had happened and restart the service. BUT it still happened at 8:49, which made me think it was something within the library.

A couple of days ago I was trying to force it to fail, so disabled NTP and manually set the machine clock. bwa kept on running. I reinstated the NTP client, set the clock back and it's been running perfectly ever since. Something weird is going on so I think I'll just start frresh with a new VM.

Anyway, sorry for the noise and thanks again for this software!

closed 2022-07-20 by fastfourier666