ccutrer/balboa_worldwide_app #69: Celsius / Fahrenheit

bbergshaven, 2022-12-17:
Having some problems changing from celsius to fahrenheit.

I see this in the logs:

`Dec 17 18:05:22 raspberrypi systemd[1]: Started Balboa Spa MQTT Bridge.
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]: /var/lib/gems/2.7.0/gems/balboa_worldwide_app-2.1.2/lib/bwa/messages/set_temperature_scale.rb:22:in `[]=': can't modify frozen String: "\\u0001\\u0000" (FrozenError)
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/balboa_worldwide_app-2.1.2/lib/bwa/messages/set_temperature_scale.rb:22:in `serialize'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/balboa_worldwide_app-2.1.2/lib/bwa/client.rb:108:in `send_message'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/balboa_worldwide_app-2.1.2/lib/bwa/client.rb:243:in `temperature_scale='
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/balboa_worldwide_app-2.1.2/exe/bwa_mqtt_bridge:194:in `block (2 levels) in publish_basic_attributes'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/homie-mqtt-1.6.3/lib/mqtt/homie/property.rb:175:in `set'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/homie-mqtt-1.6.3/lib/mqtt/homie/device.rb:118:in `block (3 levels) in publish'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/mqtt-ccutrer-1.0.3/lib/mqtt/client.rb:468:in `block in get'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/mqtt-ccutrer-1.0.3/lib/mqtt/client.rb:454:in `loop'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/mqtt-ccutrer-1.0.3/lib/mqtt/client.rb:454:in `get'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/homie-mqtt-1.6.3/lib/mqtt/homie/device.rb:107:in `block (2 levels) in publish'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]: /var/lib/gems/2.7.0/gems/balboa_worldwide_app-2.1.2/lib/bwa/client.rb:72:in `readpartial': end of file reached (EOFError)
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/balboa_worldwide_app-2.1.2/lib/bwa/client.rb:72:in `block in poll'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/balboa_worldwide_app-2.1.2/lib/bwa/client.rb:62:in `loop'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/balboa_worldwide_app-2.1.2/lib/bwa/client.rb:62:in `poll'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/balboa_worldwide_app-2.1.2/exe/bwa_mqtt_bridge:59:in `block in initialize'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/balboa_worldwide_app-2.1.2/exe/bwa_mqtt_bridge:58:in `loop'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/balboa_worldwide_app-2.1.2/exe/bwa_mqtt_bridge:58:in `initialize'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/balboa_worldwide_app-2.1.2/exe/bwa_mqtt_bridge:418:in `new'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /var/lib/gems/2.7.0/gems/balboa_worldwide_app-2.1.2/exe/bwa_mqtt_bridge:418:in `<top (required)>'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /usr/local/bin/bwa_mqtt_bridge:23:in `load'
Dec 17 18:06:49 raspberrypi bwa_mqtt_bridge[4422]:         from /usr/local/bin/bwa_mqtt_bridge:23:in `<main>'`

bbergshaven, 2022-12-19:
It does however show the current temperature as ºC. Only target temperature that is F.

[image]

ccutrer, 2022-12-19:
Fixed and release in 2.1.3. Thanks for the report.

commit 54d2492b (2022-12-19): fix set temperature scale

fixes #69