ccutrer/balboa_worldwide_app #71: Adding microsilk pump

TailoredTuna, 2023-05-07:
I have a Marquis spa with a Microsilk pump that isn't parsed by the BWA app. I'm using bwalink in a docker container. I tracked down the Toggle Item Request sent by the topside controller and also the status message byte when it is on/off. I'm looking for some assistance on adding this device to the code. My coding skills aren't too advanced and I'm getting a bit lost in the code. I added some entries to status.rb and toggle_item.rb but got some errors about a missing method. Figured I'd check here before digging deeper. Any help would be appreciated.

Status Update byte 22 is 00 when pump is off and 02 when pump is on:

D,  read: 7e 20 ff af 13 00 00 63 11 21 00 00 63 62 00 04 00 00 00 00 00 00 00 00 00 64 00 02 02 1e 00 01 74 7e
I, from spa: #<BWA::Messages::Status 5:33PM 99/100°F filter_cycles=[false, false] ready high blower=0 pumps=[0, 0, 0, 0, 0, 0] lights=[false, false] aux=[false, false]>

When turning the pump on/off via topside control it sends toggle item byte 1e:

D,  read: 7e 07 10 bf 11 1e 00 bf 7e
I, from spa: #<BWA::Messages::ToggleItem 30>

ianthepez, 2023-06-08:
Just got myself a Marquis spa with Microsilk.  This is exactly the reason I wanted to get this setup in the first place.  Microsilk takes a few min to saturate the water, and would be great to start it from inside.  I'm not much of a coder myself, but please let me know if you get it working yourself.

ccutrer, 2023-07-29:
Are you able to identify the configuration byte that indicates the presence of this pump?

TailoredTuna, 2023-07-30:
I'll have to dig back into this to get my head wrapped around it again, but if I recall correctly only the status byte changed when it was on/off, it isn't defined as a pump or anything in those config bits. 

I can gather more data if it helps. I'm not sure what else would be helpful as these are the only things I've noticed that changed.

I Appreciate your help

ccutrer, 2023-07-30:
Well, it wouldn’t change for you as the pump turns on and off. You’ll need to compare your config bytes to the docs and see if you have one byte that’s set different than others. I just don’t want to add support for this, and then have it show up for everyone else that doesn’t have that pump.

ianthepez, 2023-11-20:
Wanted to follow-up and see if there is anything I can provide that would make this possible.  I don't exactly know what I'm doing, but I'm willing to help if I can have someone give me some steps to go through.

TailoredTuna, 2023-11-20:
Sorry for not getting back on this, myself. I'm a network engineer and my coding skills are not that advanced. In the config bytes I didn't see this get identified as a pump, but I could have looked at it wrong. Let us know what we can do to help move this along.

ccutrer, 2025-05-12:
You need to post the logging from your config response. Feel free to re-open the issue if you do that.

closed 2025-05-12 by ccutrer