ccutrer/balboa_worldwide_app #73: This gem seems to work fine with non-Balboa WiFi options

supersebbo, 2023-06-17:
Just wanted to, firstly, thank @ccutrer again for all his hard work. 

Secondly to report that this code seems to work perfectly with non-Balboa R485-WiFi adaptors, in my case an Elfin-EW11. After spending many hours trying to get @jshank 's bwalink container running reliably (socat was having terrrible problems maintaining the connection), I then tried the telnetd option on the EW11 but this did not seem to be rfc2217 compliant as while I could get raw data out in a telnet session, this gem was unable to communicate with it.  So eventually in desperation I flipped the EW11 back to TCP Server mode and just entered the tcp URL into the command line for the gem.  Boom. It just works.  So far totally reliable with full reporting and control. 

So I then packaged this gem up as a HomeAssistant add-on (which required importing the full contents of the base Ruby Dockerfile as the HA base Dockerfile does not have Ruby support) which then allowed me to use the internal MQTT server of HA - which automagically did the discovery.  I'm extermely pleased!

Perhaps something to add to the Wiki?

Elfin Config: 
[image]

Home Assistant Add-on init script:
```
#!/usr/bin/with-contenv bashio
set -e

MQTT_HOST=$(bashio::services mqtt "host")
MQTT_USER=$(bashio::services mqtt "username")
MQTT_PASSWORD=$(bashio::services mqtt "password")

MQTT_URI="mqtt://${MQTT_USER}:${MQTT_PASSWORD}@${MQTT_HOST}"

echo "$MQTT_URI"

/usr/local/bundle/bin/bwa_mqtt_bridge ${MQTT_URI} tcp://10.10.10.214:9999/
```

Home Assistant Add-on Config:
```
name: "SpaLink"
description: "SpaLink Addon Dev"
version: "1.0.0"
slug: "spalink"
init: false
startup: application
services:
  - 'mqtt:need'
options:
  bridge_ip: '10.10.10.214'
  bridge_port: '9999'
arch:
  - aarch64
  - amd64
  - armhf
  - armv7
  - i386
```

The Dockerfile is just a merge of the base alpine ruby dockerfile (full copy/paste) with @jshank 's docker config added at the bottom. 

End Result:
[image]

Over the next few days I'll try to tidy this up into a publishable HA Add-On.

jshank, 2023-06-17:
This is awesome and thanks for providing the feedback! I don't think I ever tried using a non-local serial connection with bwa_mqtt_bridge. I wonder if that's always been the case or there was a code update at some point. I'll look at updating the docker container to "go direct" in the meantime but looking forward to a native ha_addon (that's the dream)! The main reason for the docker container was to avoid having to install Ruby and it's local (and sometimes finicky dependencies). Does the add-on approach work around that?

Great work!

jshank, 2023-06-17:
I rebuild my local instance to remove socat and connect directly. So far, so good. I've removed my nightly docker restart script and will let it run for a week. If all goes well, I'll publish a new container next week.

supersebbo, 2023-06-17:
Thanks!  Glad to hear it’s working for you.

I have no idea if it was part of the original code or a later update.  In the readme it suggests the app needs to know if it’s serial or WiFi because they operate differently but so far I’ve not seen evidence of this.  There may be hidden issues we’ve yet to encounter but I’m over 24 hours without an issue now. 

HomeAssistant add-ons are just docker containers with a bit of supporting meta data and some hooks into HA core.  The only thing stopping me from just using your container was that HA provides some useful tools in their base images like bashio which allows you to easily grab environment variables from HA.  In the end it was easier to add Ruby and the bridge packages to the HomeAssistant base image rather than try to add the HA tools into yours.  

The biggest advantages of running as an HA add on are:

1. you get zero-config MQTT.  If you are running a supported MQTT server already in HA then you can get the credentials for this from HA core and communicate with it over the internal HA virtual network, meaning you never have to type a thing.
2. HA has a built-in watchdog service for add-ons, so if they error for any reason it will restart them for you automatically. 
3. There is no additional overhead in running a separate Docker host as HA has one built in. 

I’ll post the complete add-on code tomorrow, if you want to get prepared then follow the initial steps here https://developers.home-assistant.io/docs/add-ons/tutorial/ to get file system access for local add-on dev (if you haven’t already).

supersebbo, 2023-06-19:
I don't want to keep hijacking @ccutrer 's issues tracker to discuss this so lets move the conversation over here:
https://github.com/supersebbo/bwa-mqtt-bridge-addon/issues/1

supersebbo, 2023-06-19:
Full installation instructions here https://github.com/supersebbo/bwa-mqtt-bridge-addon/wiki/Installation-&-Run

w4Rd3n, 2023-09-20:
> Just wanted to, firstly, thank @ccutrer again for all his hard work.
> 
> Secondly to report that this code seems to work perfectly with non-Balboa RS485-WiFi adaptors, in my case an Elfin-EW11. After spending many hours trying to get @jshank 's bwalink container running reliably (socat was having terrrible problems maintaining the connection), I then tried the telnetd option on the EW11 but this did not seem to be rfc2217 compliant as while I could get raw data out in a telnet session, this gem was unable to communicate with it. So eventually in desperation I flipped the EW11 back to TCP Server mode and just entered the tcp URL into the command line for the gem.
> 
> `/usr/local/bundle/bin/bwa_mqtt_bridge ${MQTT_URI} tcp://10.10.10.214:9999/`
> 
> Boom. It just works. So far totally reliable with full reporting and control.
> 
> So I then packaged this gem up as a HomeAssistant add-on (which required importing the full contents of the base Ruby Dockerfile as the HA base Dockerfile does not have Ruby support) which then allowed me to use the internal MQTT server of HA - which automagically did the discovery. I'm extermely pleased!
> 
> Perhaps something to add to the Wiki?
> 
> Elfin Config: [image]
> 
> Home Assistant Add-on init script:
> 
> ```
> #!/usr/bin/with-contenv bashio
> set -e
> 
> # Grab the MQTT settings from HAs internal services broker
> MQTT_HOST=$(bashio::services mqtt "host")
> MQTT_USER=$(bashio::services mqtt "username")
> MQTT_PASSWORD=$(bashio::services mqtt "password")
> 
> # Concat the MQTT URI String
> MQTT_URI="mqtt://${MQTT_USER}:${MQTT_PASSWORD}@${MQTT_HOST}"
> 
> # TO DO - pull the TCP URI from the HA Add-On Config UI
> /usr/local/bundle/bin/bwa_mqtt_bridge ${MQTT_URI} tcp://10.10.10.214:9999/
> ```
> 
> Home Assistant Add-on Config:
> 
> ```
> name: "SpaLink"
> description: "SpaLink Addon Dev"
> version: "1.0.0"
> slug: "spalink"
> init: false
> startup: application
> services:
>   - 'mqtt:need'
> options:
>   bridge_ip: '10.10.10.214'
>   bridge_port: '9999'
> arch:
>   - aarch64
>   - amd64
>   - armhf
>   - armv7
>   - i386
> ```
> 
> The Dockerfile is just a merge of the base alpine ruby dockerfile (full copy/paste) with @jshank 's docker config added at the bottom.
> 
> End Result: [image]
> 
> [image]
> 
> Over the next few days I'll try to tidy this up into a publishable HA Add-On.

Damn this is clean af.
Any way you can share how to setup the view on your samsung tablet ?
regards

supersebbo, 2023-09-21:
> Damn this is clean af. Any way you can share how to setup the view on your samsung tablet ? regards

Which part specifically?  Do you already have Home Assistant server running?  If not... you've got a bit of a learning curve coming... https://www.home-assistant.io/getting-started/ 

The tablet is a Samsung Active tab which are waterproof and can be installed outside, the only issue I have with it is occasionally overheating in direct sunlight.  It's running the Home Assistant Android client direct from the Play Store, and connected directly to my HA server. 

The code of the dashboard is here, but it's a very simple standard Lovelace view in HA.  You'd need to swap out all the sensor/entity names for how you have them configured in your HA setup.  
```
title: Garden
views:
  - title: Spa Control
    path: spa-control
    icon: mdi:hot-tub
    type: panel
    badges: []
    cards:
      - type: vertical-stack
        cards:
          - type: entity
            entity: sensor.time
          - type: horizontal-stack
            cards:
              - type: entity
                entity: sensor.outside_temperature
              - type: entity
                entity: sensor.outside_humidity
          - type: media-control
            entity: media_player.spotify_xxxxxxxxx
          - type: horizontal-stack
            cards:
              - show_name: true
                show_icon: true
                type: button
                tap_action:
                  action: toggle
                entity: light.decking_lights_light
                name: Deck Lights
                icon: mdi:lightbulb-spot
              - show_name: true
                show_icon: true
                type: button
                tap_action:
                  action: toggle
                entity: light.garden_light
                name: Flood Light
                icon: mdi:light-flood-down
              - show_name: true
                show_icon: true
                type: button
                tap_action:
                  action: toggle
                entity: light.garage_lights_light
                name: Garage Lights
                icon: mdi:garage
          - type: horizontal-stack
            cards:
              - show_name: true
                show_icon: true
                type: button
                tap_action:
                  action: toggle
                entity: light.gledopto_gl_c_008_light
                icon: mdi:hot-tub
                name: On/Off
              - show_name: true
                show_icon: false
                type: button
                tap_action:
                  action: call-service
                  service: light.turn_on
[... 117 lines cut ...]
                name: Jet Pump 2
                show_state: true
              - show_name: true
                show_icon: true
                type: button
                tap_action:
                  action: call-service
                  service: switch.toggle
                  target:
                    entity_id: switch.bwa_link_hot_tub_blower
                  data: {}
                entity: binary_sensor.bwa_link_hot_tub_circulation_pump_running
                name: Filter Pump
                show_state: true
                hold_action:
                  action: more-info
              - show_name: true
                show_icon: true
                type: button
                tap_action:
                  action: none
                entity: binary_sensor.bwa_link_hot_tub_heating
                name: Heater
                show_state: true
```

ccutrer, 2023-09-21:
Might I suggest putting the Lovelace config and addon setup on a wiki page in this repo? Then send me a PR when it's ready to link to it from the readme.

closed 2025-05-12 by ccutrer