ccutrer/balboa_worldwide_app #74: Fails to run: No such file or directory

blundell, 2023-06-18:
I'm using the Docker container (https://github.com/jshank/bwalink).

When the code runs (ie. when I execute the `docker run` command), I get this error:

```
/usr/local/bundle/gems/ccutrer-serialport-1.0.0/lib/ccutrer/serial_port.rb:7:in `initialize': No such file or directory @ rb_sysopen - /dev/hottub (Errno::ENOENT)
	from /usr/local/bundle/gems/ccutrer-serialport-1.0.0/lib/ccutrer/serial_port.rb:7:in `initialize'
	from /usr/local/bundle/gems/balboa_worldwide_app-2.1.1/lib/bwa/client.rb:49:in `new'
	from /usr/local/bundle/gems/balboa_worldwide_app-2.1.1/lib/bwa/client.rb:49:in `initialize'
	from /usr/local/bundle/gems/balboa_worldwide_app-2.1.1/exe/bwa_mqtt_bridge:413:in `new'
	from /usr/local/bundle/gems/balboa_worldwide_app-2.1.1/exe/bwa_mqtt_bridge:413:in `<top (required)>'
	from /usr/local/bundle/bin/bwa_mqtt_bridge:25:in `load'
	from /usr/local/bundle/bin/bwa_mqtt_bridge:25:in `<main>'
```

any idea?

supersebbo, 2023-06-19:
If you're running the container the permissions should be fine, more likely a network comms issue with the adapter.

blundell, 2023-06-19:
Sorry what is the adapter in what you describe?

I've got the docker container and the mqtt server running on the same machine, perhaps passing its own IP address is the wrong thing to do?

supersebbo, 2023-06-19:
The device that is communicating with your Hot Tub (serial to IP device) the typical is the Elfin-EW11 as per @jshank tutorial.

blundell, 2023-08-26:
The issue was I used an EW10 not an EW11 😆 sorry!

closed 2023-08-26 by blundell