ccutrer/balboa_worldwide_app #79: Job for bwa_mqtt_bridge.service failed

jlogan223, 2023-10-25:
I am new to coding and took on HA and other side projects like this to learn. So if I ask an obvious/stupid questions, please bare with me. haha.

I have got to the point where I need to get the Pi3 running with the Hommie bridge up and running.

I edited my bwa_mqtt_bridge.service to look like this:

```
[Unit]
Description=Balboa Spa MQTT Bridge

[Service]
User=cody
ExecStart=/usr/local/bin/bwa_mqtt_bridge mqtt://*username*:*pword*@*ipaddress*:1883 /dev/ttyUSB0
Environment="LOG_LEVEL=WARN" "LOG_VERBOSITY=0"
Restart=always
RestartSec=3s
Type=notify
NotifyAccess=main
WatchdogSec=60

[Install]
WantedBy=multi-user.target
```

I have then in my RPI Terminal I run the following:

```
sudo systemctl daemon-reload
sudo systemctl enable bwa_mqtt_bridge
sudo systemctl start bwa_mqtt_bridge
```

I then however get the following error:
```
Job for bwa_mqtt_bridge.service failed because the control process exited with error code.
See "systemctl status bwa_mqtt_bridge.service" and "journalctl -xeu bwa_mqtt_bridge.service" for details.
root@raspberrypi:/etc/systemd/system# sudo systemctl status bwa_mqtt_bridge
* bwa_mqtt_bridge.service - Balboa Spa MQTT Bridge
     Loaded: loaded (/etc/systemd/system/bwa_mqtt_bridge.service; enabled; preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Sun 2023-10-22 17:51:58 BST; 30ms ago
    Process: 8928 ExecStart=/usr/local/bin/bwa_mqtt_bridge mqtt://*username*:*pword*@*ipaddress*:1883 /dev/ttyUSB0 (code=exited, status=217/USER)
   Main PID: 8928 (code=exited, status=217/USER)
        CPU: 0

Oct 22 17:51:58 raspberrypi systemd[1]: bwa_mqtt_bridge.service: Main process exited, code=exited, status=217/USER
Oct 22 17:51:58 raspberrypi systemd[1]: bwa_mqtt_bridge.service: Failed with result 'exit-code'.
Oct 22 17:51:58 raspberrypi systemd[1]: Failed to start bwa_mqtt_bridge.service - Balboa Spa MQTT Bridge.
root@raspberrypi:/etc/systemd/system# sudo journalctl -xeu bwa_mqtt_bridge
-- The job identifier is 278955.
Oct 22 17:52:37 raspberrypi systemd[1]: bwa_mqtt_bridge.service: Main process exited, code=exited, status=217/USER
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- An ExecStart= process belonging to unit bwa_mqtt_bridge.service has exited.
-- 
-- The process' exit code is 'exited' and its exit status is 217.
Oct 22 17:52:37 raspberrypi systemd[1]: bwa_mqtt_bridge.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- The unit bwa_mqtt_bridge.service has entered the 'failed' state with result 'exit-code'.
Oct 22 17:52:37 raspberrypi systemd[1]: Failed to start bwa_mqtt_bridge.service - Balboa Spa MQTT Bridge.
-- Subject: A start job for unit bwa_mqtt_bridge.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A start job for unit bwa_mqtt_bridge.service has finished with a failure.
-- 
-- The job identifier is 278955 and the job result is failed.
lines 7478-7500/7500 (END)
-- The job identifier is 278955.
Oct 22 17:52:37 raspberrypi systemd[1]: bwa_mqtt_bridge.service: Main process exited, code=exited, status=217/USER
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- An ExecStart= process belonging to unit bwa_mqtt_bridge.service has exited.
-- 
-- The process' exit code is 'exited' and its exit status is 217.
Oct 22 17:52:37 raspberrypi systemd[1]: bwa_mqtt_bridge.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- The unit bwa_mqtt_bridge.service has entered the 'failed' state with result 'exit-code'.
Oct 22 17:52:37 raspberrypi systemd[1]: Failed to start bwa_mqtt_bridge.service - Balboa Spa MQTT Bridge.
-- Subject: A start job for unit bwa_mqtt_bridge.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A start job for unit bwa_mqtt_bridge.service has finished with a failure.
-- 
-- The job identifier is 278955 and the job result is failed.
```

Are you able to give me some idea of what I am doing wrong?

I have spent a few days looking through Google for possible hints l, the best I can get is that my user credentials could be wrong, but I have deleted and re-entered them so many times, it can't be a typo. Plus I have checked existing users with MQTT access ) that work on other devices, created new users and confirmed they work on other devices too.

I have tried wiping the pi 3b+, and with a fresh install I get the same issue again.

Thanks for any help you can give me. I really want to get this working as it looks so cool!

ianthepez, 2023-11-21:
Mine looks like this in case it helps.

[Unit]
Description=Balboa Spa MQTT Bridge

[Service]
User=****
ExecStart=/usr/local/bin/bwa_mqtt_bridge mqtt://****:****@192.168.4.160:1883 tcp://192.168.4.218:8899
Environment="LOG_LEVEL=WARN" "LOG_VERBOSITY=0"
Restart=always
RestartSec=3s
Type=notify
NotifyAccess=main
WatchdogSec=60

[Install]
WantedBy=multi-user.target

StKi1da, 2024-01-08:
@jlogan223 

Did you get this resolved - I have the same error and can't figure it out?

jlogan223, 2024-01-08:
Yes, I was trying to set it up without the module connected to my Hottub, hoping I could make sure everything worked before going outside and taking the panels off. Turns out that ( now obvious,  hindsight), it only works once it is connected to the hotub so it has some data to transmit. Plug it up to the hottub and then see if it starts working or bot.

StKi1da, 2024-01-08:
Ahh. Ditto. So the file details modified as you posted originally work fine
once connected? Did you need to modify the user=cody line?

Many thanks for replying so quickly

JPCClout, 2024-02-09:
Hi, I get this "bwa_mqtt_bridge.service: Failed to determine user credentials" Error too. Tried many variations but stuck there. MQTT set to allow_anonymous true, with no username/password in config, still same error. 
Can you use the same RasPi 4 for the bridge AND broker? How would you then write the mqqt address in the config file? "mqtt://localhost:1883" ? 
Anyone used homebridge and mqttthing plugin? Any success?
Thanks

StKi1da, 2024-03-31:
> Hi, I get this "bwa_mqtt_bridge.service: Failed to determine user credentials" Error too. Tried many variations but stuck there. MQTT set to allow_anonymous true, with no username/password in [...]

How did you get on? I’m struggling to figure out how to configure the bwa_mqtt_bridge.service file correctly

ccutrer, 2024-04-01:
That error sounds like it's coming from systems, not the Ruby gem. are you sure the user (not the MQTT user - the Linux user) you entered into the service file is valid?

StKi1da, 2024-04-01:
Thanks for the quick reply. Apologies as I am new to HA and each
integration is a new learning for me! I'm guessing I have not got the set
up of MQTT / HA and/or the client side quite correct.

So to be clear - I followed the Mosquitto add on in HA and the integration
appears- and I created a new user in HA (mqtt-user)  - which is active.
It's that user and password I am putting into the service file.

ExecStart=/usr/local/bin/bwa_mqtt_bridge ***@***.***_address
of HA:1883 /dev/ttyUSB0

Is this not correct?

Many thanks for your support!

ccutrer, 2024-04-01:
It may or may not be correct - but is not the line I'm referring to at all. The line above it -- `User=****` -- is what I'm referring to.

StKi1da, 2024-04-01:
ok. I have tried User= with both the client side username (the username I
use to log in with ssh to the raspberry pi connected to the hot tub) and
the HA username and the mqtt user i set up for the broker. While the ssh
username will have root permissions (the user appears on the sudo group) -
Do I need to enable root by default? sorry of that's dumb question!

ccutrer, 2024-04-01:
Yes, you should use the ssh username. Just because a user is in the sudoers file, doesn't mean they have root permissions. Unless it's set up as NOPASSWD, a process can't escalate to root.

StKi1da, 2024-04-01:
Thanks for your patience!

So I have reinstalled Rasbian and Ruby-dev and then the gem which seemed to
go ok. I amended sudoers file to include user1 as NOPASSWD so it no reads
as below

root    ALL=(ALL:ALL) ALL

user1 ALL=(ALL:ALL) NOPASSWD: ALL

and adjusted the service file:

*[Unit]*
>
> Description=Balboa Spa MQTT Bridge
>
>
> *[Service]*
>
> User=user1
>
> ExecStart=/usr/local/bin/bwa_mqtt_bridge mqtt://*mqttuser*:*pass*:1883
> /dev/ttyUSB0
>
> Environment="LOG_LEVEL=WARN" "LOG_VERBOSITY=0"
>
> Restart=always
>
> RestartSec=3s
>
> Type=notify
>
> NotifyAccess=main
>
> WatchdogSec=60
>
>
> *[Install]*
>
> WantedBy=multi-user.target
>

 but still getting an error but seems different now!  "exit code 1"

Any suggestions?

***@***.****:*~ $* sudo systemctl start bwa_mqtt_bridge

Job for bwa_mqtt_bridge.service failed because the control process exited
with error code.

See "systemctl status bwa_mqtt_bridge.service" and "journalctl -xe" for
details.

***@***.****:*~ $* sudo systemctl status bwa_mqtt_bridge.service

● bwa_mqtt_bridge.service - Balboa Spa MQTT Bridge

     Loaded: loaded (/etc/systemd/system/bwa_mqtt_bridge.service; enabled;
vendor preset: enabled)

     Active: activating (start) since Mon 2024-04-01 20:26:13 BST; 578ms ago

   Main PID: 7125 (bwa_mqtt_bridge)

      Tasks: 1 (limit: 414)

        CPU: 203ms

     CGroup: /system.slice/bwa_mqtt_bridge.service

             └─7125 /usr/bin/ruby2.7 /usr/local/bin/bwa_mqtt_bridge
mqtt://mqtt:pass:1883 /dev/ttyUSB0

Apr 01 20:26:13 raspberrypi systemd[1]: Starting Balboa Spa MQTT Bridge...

***@***.****:*~ $*

The process' exit code is 'exited' and its exit status is 1.

Apr 01 20:28:53 raspberrypi systemd[1]: *bwa_mqtt_bridge.service: Failed
with result 'exit-code'.*

░░ Subject: Unit failed

░░ Defined-By: systemd

░░ Support: https://www.debian.org/support

░░

░░ The unit bwa_mqtt_bridge.service has entered the 'failed' state with
result 'exit-code'.

Apr 01 20:28:53 raspberrypi systemd[1]: *Failed to start Balboa Spa MQTT
Bridge.*

░░ Subject: A start job for unit bwa_mqtt_bridge.service has failed

░░ Defined-By: systemd

░░ Support: https://www.debian.org/support

░░

░░ A start job for unit bwa_mqtt_bridge.service has finished with a failure.

░░

░░ The job identifier is 37624 and the job result is failed.

Apr 01 20:28:53 raspberrypi systemd[1]: *bwa_mqtt_bridge.service: Consumed
7.278s CPU time.*

░░ Subject: Resources consumed by unit runtime

░░ Defined-By: systemd

░░ Support: https://www.debian.org/support

░░

░░ The unit bwa_mqtt_bridge.service completed and consumed the indicated
resources.

Apr 01 20:28:53 raspberrypi sudo[7160]: *   user1 : TTY=pts/0 ;
PWD=/home/user1 ; USER=root ; COMMAND=/usr/bin/journalct*>

Apr 01 20:28:53 raspberrypi sudo[7160]: pam_unix(sudo:session): session
opened for user root(uid=0) by user1(uid=1000)

lines 3288-3310/3310 (END)

carldunso, 2024-05-03:
Hi! Did you manage to solve this? I think I am in the exact same position as you. The installation and config setup went fine but the service wont start and gives the same errors.

StKi1da, 2024-05-03:
No. I have stopped trying to get it to work for the moment but will give it
another go in a few weeks time

jlogan223, 2024-05-03:
> Hi! Did you manage to solve this? I think I am in the exact same position as you. The installation and config setup went fine but the service wont start and gives the same errors.

Is the unit connected to your hot tub?

carldunso, 2024-05-03:
Which rs485 adapter do you have? Maybe it is there the problem is. I ordered two other ones just to try.

I will post here if I get any progress with the problem. Maybe you can do the same?

carldunso, 2024-05-03:
>Is the unit connected to your hot tub?

I have tried both ways. Connected and not. Get the same error.

jlogan223, 2024-05-03:
My error only occurred when not connected. Ensure you have it connected properly and wired up correctly. If that's not the problem, then I don't know. Be careful with MQTT user name and passwords , it doesn't like special characters

carldunso, 2024-05-03:
That make me think that the problem is my RS485 to USB adapter. I will have to wait until next week when the two other arrives! Thanks 😊

carldunso, 2024-05-03:
These two:

[image]

[image]

StKi1da, 2024-05-03:
> These two:
>
> Screenshot_20240503_222225_Chrome.jpg (view on web)
>
> Screenshot_20240503_222236_Chrome.jpg (view on web)

I originally tried the op one of your links.

Just ordered this one

carldunso, 2024-05-06:
I can confirm that the service worked with this USB-adapter:
https://www.amazon.se/dp/B07B416CPK?psc=1&ref=ppx_yo2ov_dt_b_product_details

Just followed the guide. Adapter showed as ttyUSB0. Modified the config file, did a daemon reload and started bwa service. Worked instantly!

closed 2025-05-12 by ccutrer