ccutrer/balboa_worldwide_app #85: Compatibility wth Balboa GS500Z and CalSpas 407T

smitterer, 2024-06-09:
Hi!

Thanks for creating this project.

We have a Balboa GS500Z connected to a CalSpas 407T panel.

The PCB has RJ connectors and one 8-pin Molex.
The documentation shows the pcb: https://www.balboawatergroup.com/getdoc.cfm?id=2150

Any idea if this project would work, either via RJ or Molex?

Thanks!

smitterer, 2024-06-15:
Ok so the specific version of the GS500Z is a CS6215R1A which is some version for Cal Spas 

Any idea if that would work, especially by using a an elfin ew-11?

Thanks

bggardner, 2024-06-15:
My guess is no, based on #14 and #57.

smitterer, 2024-06-15:
Just checked again (just received it, bought it used and oversaw that) and there is only the RJ connector, see:
https://photos.app.goo.gl/NoeShParJBgJhQFB9
https://photos.app.goo.gl/DKkjtgdfRPYoWXpB6

closed 2025-05-12 by ccutrer