ccutrer/balboa_worldwide_app #86: Version 2.1.5 does not work with Balboa spa BP2100G1

TheNetStriker, 2024-07-13:
I've just tested the new version 2.1.5 with my Balboa spa BP2100G1, but sadly no values are reported via MQTT.

The log output only shows the following:

I,   to spa: #<BWA::Messages::ConfigurationRequest>
I,   to spa: #<BWA::Messages::ControlConfigurationRequest 3>

Before I was using version 1.2.4 which reported all the values. I only had to fix the pump values: https://github.com/TheNetStriker/balboa_worldwide_app/tree/PumpSingleSpeedFix

Is there anything I could provide to help find the problem?

ccutrer, 2024-07-14:
If you go back to 1.2.4, does it still work?

devotee101, 2024-09-15:
I'm seeing the same issue.

I'm running the app as part of https://github.com/jshank/bwalink - Release 2021.12 which uses v1.3.0 of the app works fine, but release 2021.12.1 or later which use the v2.x.x app just hang at ControlConfigurationRequest.

Here's the debug output from the v2.x.x app:
```
I,   to spa: #<BWA::Messages::ConfigurationRequest>
D, wrote: 7e 05 0a bf 04 77 7e
D, discarding invalid data prior to message de a0 df 15 e2 e9 03 e0 03 f5 df 81 be d4 f0 a0 df 17 e2 49 03 a0 df 15 e6 e7 03 e0 a0 df 15 e2 e9 03 e0
D,  read: 7e 05 0a bf 04 77 7e
I, from spa: #<BWA::Messages::ConfigurationRequest>
D, discarding invalid data prior to message 03 f5 df 81 be d4 fe a0 df 17 e2 49 03 03 f5 5b 81 be de a0 df 15 e2 e9 03 e0 03 f5 df 81 be d4 f0 a0 df 17 e2 49 03 03 f5 5b 81 be de a0 df 15 e2 e9 03 e0 03 f5 df 81 be d4 f0 a0 df 17 e2 49 03 03 f5 5b 81 be de f0 a0 df 15 e2 e9 03 e0 03 f5 df 81 be d4 f0 a0 df 17 e2 49 03 03 f5 5b 81 be de a0 df 15 e2 e9 03 e0 03 bf 01 28 76 ff 67 eb ab ff ff 6b 6b f9 e7 ff ff fb ff ff ff ff ff ff 67 ff ff df c3 ff ff 39 20 a0 df 17 e6 47 03 e0 a0 df 17 e2 49 03 a0 df 15 e6 e7 03 f0 a0 df 15 e2 e9 03 e0
D,  read: 7e 08 0a bf 22 01 00 00 34 7e
I, from spa: #<BWA::Messages::ControlConfigurationRequest 3>
```

Contrast with v1.3.0:
```
I,   to spa: #<BWA::Messages::ConfigurationRequest>
D,   to mqtt: $homie: 4.0.0
D,   to mqtt: $name: BWA Spa
D,   to mqtt: $state: init
D,   to mqtt: $nodes: spa
D,   to mqtt: spa/$name: BWA Spa
D,   to mqtt: spa/$type: spa
D,   to mqtt: spa/$properties: hold,priming,heatingmode,temperaturescale,24htime,heating,temperaturerange,currenttemperature,settemperature,filter1,filter2
D,   to mqtt: spa/hold/$name: In Hold mode
D,   to mqtt: spa/hold/$datatype: boolean
D,   to mqtt: spa/hold/$settable: true
D,   to mqtt: spa/priming/$name: Is the pump priming
D,   to mqtt: spa/priming/$datatype: boolean
D,   to mqtt: spa/heatingmode/$name: Current heating mode
D,   to mqtt: spa/heatingmode/$datatype: enum
D,   to mqtt: spa/heatingmode/$format: ready,rest,ready_in_rest
D,   to mqtt: spa/heatingmode/$settable: true
D,   to mqtt: spa/temperaturescale/$name: Temperature scale
D,   to mqtt: spa/temperaturescale/$datatype: enum
D,   to mqtt: spa/temperaturescale/$format: fahrenheit,celsius
D,   to mqtt: spa/temperaturescale/$settable: true
D,   to mqtt: spa/24htime/$name: Clock is 24 hour time
D,   to mqtt: spa/24htime/$datatype: boolean
D,   to mqtt: spa/24htime/$settable: true
D,   to mqtt: spa/heating/$name: Heater is currently running
D,   to mqtt: spa/heating/$datatype: boolean
D,   to mqtt: spa/temperaturerange/$name: Current temperature range
D,   to mqtt: spa/temperaturerange/$datatype: enum
D,   to mqtt: spa/temperaturerange/$format: high,low
D,   to mqtt: spa/temperaturerange/$settable: true
D,   to mqtt: spa/currenttemperature/$name: Current temperature
D,   to mqtt: spa/currenttemperature/$datatype: float
D,   to mqtt: spa/settemperature/$name: Set Temperature
D,   to mqtt: spa/settemperature/$datatype: float
D,   to mqtt: spa/settemperature/$settable: true
D,   to mqtt: spa/filter1/$name: Filter cycle 1 is currently running
D,   to mqtt: spa/filter1/$datatype: boolean
D,   to mqtt: spa/filter2/$name: Filter cycle 2 is currently running
D,   to mqtt: spa/filter2/$datatype: boolean
D,   to mqtt: spa/filter1hour/$name: Filter Cycle 1 Start Hour
D,   to mqtt: spa/filter1hour/$datatype: integer
D,   to mqtt: spa/filter1hour/$format: 0:24
D,   to mqtt: spa/filter1hour/$settable: true
D,   to mqtt: spa/filter1minute/$name: Filter Cycle 1 Start Minutes
D,   to mqtt: spa/filter1minute/$datatype: integer
D,   to mqtt: spa/filter1minute/$format: 0:59
D,   to mqtt: spa/filter1minute/$settable: true
D,   to mqtt: spa/filter1durationhours/$name: Filter Cycle 1 Duration Hours
D,   to mqtt: spa/filter1durationhours/$datatype: integer
D,   to mqtt: spa/filter1durationhours/$format: 0:24
D,   to mqtt: spa/filter1durationhours/$settable: true
D,   to mqtt: spa/filter1durationminutes/$name: Filter Cycle 1 Duration Minutes
D,   to mqtt: spa/filter1durationminutes/$datatype: integer
D,   to mqtt: spa/filter1durationminutes/$format: 0:59
D,   to mqtt: spa/filter1durationminutes/$settable: true
D,   to mqtt: spa/filter2enabled/$name: Filter Cycle 2 Enabled
D,   to mqtt: spa/filter2enabled/$datatype: boolean
D,   to mqtt: spa/filter2enabled/$settable: true
D,   to mqtt: spa/filter2hour/$name: Filter Cycle 2 Start Hour
D,   to mqtt: spa/filter2hour/$datatype: integer
D,   to mqtt: spa/filter2hour/$format: 0:24
D,   to mqtt: spa/filter2hour/$settable: true
D,   to mqtt: spa/filter2minute/$name: Filter Cycle 2 Start Minutes
D,   to mqtt: spa/filter2minute/$datatype: integer
D,   to mqtt: spa/filter2minute/$format: 0:59
D,   to mqtt: spa/filter2minute/$settable: true
D,   to mqtt: spa/filter2durationhours/$name: Filter Cycle 2 Duration Hours
D,   to mqtt: spa/filter2durationhours/$datatype: integer
D,   to mqtt: spa/filter2durationhours/$format: 0:24
D,   to mqtt: spa/filter2durationhours/$settable: true
D,   to mqtt: spa/filter2durationminutes/$name: Filter Cycle 2 Duration Minutes
D,   to mqtt: spa/filter2durationminutes/$datatype: integer
D,   to mqtt: spa/filter2durationminutes/$format: 0:59
D,   to mqtt: spa/filter2durationminutes/$settable: true
D,   to mqtt: spa/$properties: hold,priming,heatingmode,temperaturescale,24htime,heating,temperaturerange,currenttemperature,settemperature,filter1,filter2,filter1hour,filter1minute,filter1durationhours,filter1durationminutes,filter2enabled,filter2hour,filter2minute,filter2durationhours,filter2durationminutes
W, Balboa MQTT Bridge running (version 1.3.0)
```

Happy to help out with some debugging if you need any more info.

TheNetStriker, 2025-01-06:
@ccutrer Yes it worked with version 1.2.4. I just tested with the latest version 1.2.7 and it seems to work again. I just had to re-add the homie thing in OpenHab and change some item links.

@devotee101 Does it work for you too in version 1.2.7? Then we could close this issue.

devotee101, 2025-01-08:
I think my issue was related to hardware. Happy to close this.

closed 2025-01-08 by ccutrer