ccutrer/balboa_worldwide_app #90: balboa worldwide app access and configuration

pierafal, 2025-01-03:
I've finally connected my balboa and looks like autodiscovery works also in Domoticz, i can see 16 devices named ESP82SPA_xx. And that is really grat, i can set the temp, turn on the lights or pumps..
[image]

But, i was expecting some kind of webpage when logging to the Wemos device using it's IP.

When i try to access it through web browser (Chrome), by entering it's IP address I can only see:
"Not found: /"

When i try IPaddress/update i can see this menu:
[image]

How can I access the device and configure it?
I'd  need to change the number of pumps to 3 (by default i have "Hot tube jet1" and "Hot tube jet2"

ccutrer, 2025-01-03:
I'm not sure which project you're using. This project is Ruby, and does not run on ESP/Wemos devices.  If you're just using the Wemos device as a remote RS485 serial port with this project... this project has no web page. All configuration is done automatically by communicating with the spa.

pierafal, 2025-01-04:
is there a way to change number of jet pomps to 3?
Is there a way to change the lowest temp to be set? Physically i can set 10degC but this tool does allow to set between 27 and 40 only,

closed 2025-05-12 by ccutrer