ccutrer/balboa_worldwide_app #91: Unrecognized Data Length

jfdid, 2025-01-05:
Hi folks, I'm actually using this through https://github.com/jshank/bwalink but it looks like this may be an issue on this end. 
Can you kindly have a look at let me know?
``
I,   to spa: #<BWA::Messages::ConfigurationRequest>

/usr/local/bundle/gems/balboa_worldwide_app-2.1.5/lib/bwa/message.rb:78:in `parse': Unrecognized data length (28) for message BWA::Messages::Status (BWA::InvalidMessage)

	from /usr/local/bundle/gems/balboa_worldwide_app-2.1.5/lib/bwa/client.rb:65:in `block in poll'

	from <internal:kernel>:187:in `loop'

	from /usr/local/bundle/gems/balboa_worldwide_app-2.1.5/lib/bwa/client.rb:64:in `poll'

	from /usr/local/bundle/gems/balboa_worldwide_app-2.1.5/exe/bwa_mqtt_bridge:33:in `block in initialize'

	from <internal:kernel>:187:in `loop'

	from /usr/local/bundle/gems/balboa_worldwide_app-2.1.5/exe/bwa_mqtt_bridge:32:in `initialize'

	from /usr/local/bundle/gems/balboa_worldwide_app-2.1.5/exe/bwa_mqtt_bridge:421:in `new'

	from /usr/local/bundle/gems/balboa_worldwide_app-2.1.5/exe/bwa_mqtt_bridge:421:in `<top (required)>'

	from /usr/local/bundle/bin/bwa_mqtt_bridge:25:in `load'

	from /usr/local/bundle/bin/bwa_mqtt_bridge:25:in `<main>'
``

jfdid, 2025-01-05:
I've worked on this for a few hours, abandoning the jshank Docker method and using this Ruby Gem directly.
This is what it seems to be hanging up on:

`I,   to spa: #<BWA::Messages::ConfigurationRequest>

D, wrote: 7e 05 0a bf 04 77 7e

D, discarding invalid data prior to message 7e

D,  read: 7e 1c ff af 13 14 00 66 16 0b 00 00 66 66 00 0c 01 00 00 00 00 00 00 00 00 66 01 00 b2 7e

/usr/local/bundle/gems/balboa_worldwide_app-2.1.5/lib/bwa/message.rb:78:in `parse': Unrecognized data length (28) for message BWA::Messages::Status (BWA::InvalidMessage)`
Although I saw `MESSAGE_LENGTH = (24..32).freeze` in the code somewhere.  
Any help would be greatly appreciated!

Model: BP20MSSH
Top panel: TP800

ccutrer, 2025-01-06:
The `MESSAGE_LENGTH` is a bit confusing, because those constants are exclusive of headers, while the length reported in that message is inclusive of them. But... that last byte isn't really used, so I've loosened that. restriction to be 23..32. Can you try v2.1.8?

jfdid, 2025-01-06:
AWESOME (SPAWESOME?)  This worked perfectly.  Thank you very much @ccutrer, I was beginning to think this board wasn't compatible.  
I see the bottom of the Wiki has some examples of data - do you want another one from my tub?  I don't know how to generate that data, but would be happy to do so if you tell me how.
Do you have a donate/buy me a beer link?  I'd be happy to send over a token of appreciation for this project.

ccutrer, 2025-01-06:
You might ask @bggardner if he wants any more details in the examples sections. He's mostly been the one maintaining the wiki. I don't really have a donate thing, but if you really want to I do have [Venmo](https://account.venmo.com/u/ccutrer) or [PayPal](https://paypal.me/ccutrer).

jfdid, 2025-01-06:
Sent via PayPal, thanks again!
You should post a donate link on your readme.

ccutrer, 2025-01-06:
Thanks. I've added a few links to the GitHub sponsorship thingy

bggardner, 2025-01-06:
If you could post your `HARDWARE` and `INFORMATION` sections in the Example section, we've already got one of your status messages above.  A byte dump of the Information Response message would be great too.  There isn't a script to generate the data, it's all manual entry.

TBH, I'm surprised this issue wasn't discovered earlier, as my Status Message has the same number of bytes.  Maybe only you and I have the older spas.  However, I use [pybalboa](https://github.com/garbled1/pybalboa) instead of this Ruby gem...I just contribute to the Wiki.