ccutrer/balboa_worldwide_app #92: Spa fault codes and reset?

Anttibay, 2025-01-08:
Hello, 

I highly appreciate the app and I am using it through Home Assistant BWA Link. 

Sometimes during winter times, the spa has either a water flow fault or suspected dry heater fault, and that is only visible trough the spa's physical control panel and the screen. Will fault related sensors and controls be part of future releases perhaps? I understood from the wiki that they have quite nicely been reverse engineered?

Thanks!

Anttibay, 2025-01-13:
@ccutrer thanks for the answer! 

I tried to interpret from the code that something regarding clear notifications has been implemented and if I physically go to the spa and clear the notifications, BWALink addon logs show:

I, from spa: #<BWA::Messages::ToggleItem clear_notification>
D,  read: 7e 07 10 bf 11 04 00 6a 7e

But it appears the app cannot discover this feature. I am also so far from dev expert that I cannot determine if a MQTT topic has been defined in the code. So if there is, what would be MQTT topic and payload for the clear_notification toggle?

Anttibay, 2025-01-14:
Answering to myself after further digging the code that publishing following MQTT message results in clearing the nofication:

Topic: homie/bwa/spa/command/set
Payload: clear_notification

I did not want to fiddle with the code as I don't have the required expertise, but instead I just send that MQTT message manually over Home Assistant button.

Havent figure out yet if the fault log request can be sent and interpreted somehow.

ccutrer, 2025-01-15:
I just released version 2.2.0 that exposes the Clear Notification functionality as a button to Home Assistant. Exposing current faults would be a separate feature.

Anttibay, 2025-01-16:
Superb! Weird however, that this also changed the maximum value of filter cycle 1 to 120 minutes instead of previous much bigger max value and target water temperature max is now 100 degrees (celsius) instead of I guess something like 40. Filter 2 cycle does now not seem to have duration, but I cannor recall if it was there earlier or not.