ccutrer/balboa_worldwide_app #94: Heating sensor backwards compatibility

KNXBroker, 2025-02-02:
Hi, thank you for working again at the home assistant integration! Is there any chance to introduce the heating sensor again for backwards compatibility? Using the water heater component does not really work in my case (ranges do not match) and I am not really a fan of the wording 'eletric' if the pool is heating. I would be very grateful if you could add the old heating sensor again. Thank you 😄

KNXBroker, 2025-02-02:
PS: Are you sure that assigning the heating sensor to operation mode is correct? Not sure to be honest. Best regards

tmason818, 2025-02-16:
> Hi, thank you for working again at the home assistant integration! Is there any chance to introduce the heating sensor again for backwards compatibility? Using the water heater component does not [...]

Yikes yeah the min/max of 110/140 threw some of my automations and graphs into chaos.

I can't seem to even find a way to chart **History** for heating anymore? I can find the entry for "off" for not heating and "electric" for heating respectively but I can't find history information????

ccutrer, 2025-05-12:
I'm not sure what you're talking about 110/140. I can clearly see my last discovery message setting min/max to 50-106 °F. Perhaps a Home Assistant bug? I have no plans to switch back to a `sensor` component when the current temperature is exposed as a water heater. Especially since I can no longer do more than trivial development on this gem myself (see the new note in the readme). You're always welcome to manually configure an MQTT Sensor yourself in Home Assistant, or specifically install an older version of this gem before Water Heater was supported.

closed 2025-05-12 by ccutrer