ccutrer/balboa_worldwide_app #97: Support for Sundance 880 2012

suaveolent, 2025-07-01:
Hello everyone,

thanks for this awesome project. I have been trying to decode the protocol of my Sundance Spa 880 - 2012 for quite a while.
Unfortunately, the board used does not expose a dedicated RS-485 port, but analysing the ports suggests it might use RS-485 too.

For reference, I'm using this display: Sundance Spas Control Panel LCD P/N 6600-882 (e.g. https://pool-spa-supplies.com/sundance-spas-topside-control-880-panel-2-pump-2012-6600-882) and attached my probes to the ribbon cable between the controller and the display.

The board is similar to, 
[image] where the LCD Panel plugs into J7.

Here are the voltages I measured for the PINs on J7.

|     | D6     | D4     | D2     | D0     |
|-----|--------|--------|--------|--------|
| GND | 4,95V  | 4,95V  | 12,23V | 12,23V |
| GND | 0,05V  | 2,74V  | 4,97V  | 4,98V  |
|     | D7     | D5     | D3     | D1     |

I also connected my logic analyzer and a RS-422/RS-485 adapter to the board to check if it has the same protocol and attached the dumps. Can you guys maybe help out and confirm/refute whether I am on the right track here?

[rs_422_ra_d4_rb_d5.txt](https://github.com/user-attachments/files/20999088/rs_422_ra_d4_rb_d5.txt)
[rs_422_ra_d4_rb_d6.txt](https://github.com/user-attachments/files/20999094/rs_422_ra_d4_rb_d6.txt)
[rs_422_ra_d5_rb_d3.txt](https://github.com/user-attachments/files/20999097/rs_422_ra_d5_rb_d3.txt)
[rs_422_ra_d5_rb_d4.txt](https://github.com/user-attachments/files/20999092/rs_422_ra_d5_rb_d4.txt)
[rs_422_ra_d6_rb_d4.txt](https://github.com/user-attachments/files/20999090/rs_422_ra_d6_rb_d4.txt)
[rs_485_a_d3_b_d5.txt](https://github.com/user-attachments/files/20999096/rs_485_a_d3_b_d5.txt)
[rs_485_a_d4_b_d6.txt](https://github.com/user-attachments/files/20999091/rs_485_a_d4_b_d6.txt)
[rs_485_a_d5_b_d3.txt](https://github.com/user-attachments/files/20999093/rs_485_a_d5_b_d3.txt)
[rs_485_a_d5_b_d4.txt](https://github.com/user-attachments/files/20999095/rs_485_a_d5_b_d4.txt)
[rs_485_a_d6_b_d4.txt](https://github.com/user-attachments/files/20999089/rs_485_a_d6_b_d4.txt)

lmaertin, 2026-08-17:
HI @suaveolent, is it working for you? Do you had some luck to find patterns in the communication? I tried many constellations now - no proper decoding possible :-(

lmaertin, 2026-08-30:
@suaveolent In all of my measurements there is so much noise data. I think sundance encrypted the communication. Any progress?

suaveolent, 2026-09-03:
unfortunately I still did not have time to setup everything. I will come back to you once I have properly setup and wired everything.

lmaertin, 2026-11-03:
I consider to change the logic and build an interactive script instructing you to press specific buttons on the panel. At the same time a continuously signal recording is done. The whole thing is controlled by the script... 

Afterwards the sequence is checked for reoccurring patterns...

This inspired me: https://youtu.be/GlQO8fLgHi4

pete-jorgensen, 2026-11-13:
@lmaertin @suaveolent, sorry for missing the ping in July.

The Sundance boards I know of do scramble the panel bus, but it is a simple XOR fold, not real encryption. Over RS485 at 115200 8N1, each status payload is sent as byte pairs: the value is `a ^ b ^ 1`. A press is one `0xCC` frame where the button code is byte 5 XOR byte 6. The decode, a field map and how it was worked out are in [DECODING.md](https://github.com/pete-jorgensen/SundanceHotTub2MQTT/blob/master/DECODING.md). It now covers the 780, the 680 and a Montclair.

Your interactive script is the right idea. The menus here were decoded that way: one known press at a time, with the frames recorded around it (`tools/panel_walk.py` in that repo).

One thing to check first. These boards frame every message with `7E ... 7E` on a differential RS485 pair. With a logic analyser on J7, single-ended and possibly at 12 V levels, it can look like noise. A cheap RS485-to-USB or RS485-to-TCP adapter on the A/B pair at 115200 should show the `7E` frames if the 880 uses the same bus. If it does, I'm happy to look at a capture.

(Written with help from an AI assistant.)

lmaertin, 2026-11-13:
Hi @pete-jorgensen 

Thanks for your detailed answer. I hope your theory is valid for the 880 as well. 

So far I was not able to detect such a pattern. I have not yet connected a rs485 dongle. I have no idea which one of the 4 signal lines carry the relevant an and b line...

Is that the same with the 780?

lmaertin, 2026-11-13:
Not sure if the generation using Balboa protocol already.

Do have an images of the connector on your spa?

pete-jorgensen, 2026-11-13:
@lmaertin I do not think the 880 board uses the same bus as the 780, from what you describe.

On my 780 the panel bus is a plain RS485 pair: a UART at 115200 8N1, with frames that start and end with `7E`. An RS485-to-TCP adapter on that pair reads it with no clock line. What you found on J7 is different: a clocked, latch-delimited frame of about 96 bits that repeats as a display scan. That looks like a synchronous shift-register bus that drives the LCD segments, not a UART. If that is right, an RS485 dongle will show only noise, and the XOR fold from my last comment will not apply.

Two checks could settle it:
- Look for a steady clock: one J7 line that toggles at a fixed rate all the time. A UART has no clock line.
- Look for the `7E` delimiter in a capture decoded as a UART at 115200. My 780 shows it in every frame, and suaveolent's 2025 dumps did not.

If it is a segment bus, the way forward is probably to map the bits to LCD segments, using your captures with known display text ("12:00", "29.9C"). That is closer to what you already started than to my decoder.

Here is the panel end of my 780 board. The panel bus has three 4-pin (2×2) sockets marked MAIN PANEL: J13, J14 and J15. The panel cable uses J13; J14 and J15 are free in this photo. So the 780 gives the bus its own connectors, where your 880 has the 2×5 J7 ribbon. I did not record the pin order on mine, so I cannot tell you which of the four pins are A and B.

[image]

There is also a wider view of the board's top edge in the README: https://github.com/pete-jorgensen/SundanceHotTub2MQTT#the-hardware-it-runs-on

This reply was written with AI assistance (Claude Code).